moveit/moveit #1136: cleanup chomp stuff

rhaschke, 2018-10-25:
Releasing Kinetic + Melodic was a nightmare due to chomp, which now is scattered around the whole repository creating many inter-dependencies between release packages.
I suggest to move all chomp-related source into a single folder, namely `moveit_planners/chomp`, which is then released as a meta-package `moveit_planners_chomp`.
- [x] **moveit_planners/chomp**
- [x] **moveit_planners/chomp**/chomp_interface
- [x] **moveit_planners/chomp**/chomp_motion_planner
- [ ] moveit_ros/planning/planning_request_adapter_plugins/src/chomp_optimizer_adapter.cpp
     -> moveit_planners/chomp/chomp_optimizer_adapter
- [ ] moveit_experimental/collision_distance_field
     -> moveit_planners/chomp/collision_distance_field
- Are there more dependencies from `moveit_experimental`?

Then, we should get rid of `moveit_experimental` dependencies in those released packages and finally mark `moveit_experimental` to be not released again, i.e. reverting https://github.com/ros-gbp/moveit-release/pull/6.

@raghavendersahdev If we agree on this approach, could you tackle this please?

v4hn, 2018-10-25:
I fully agree. I was not involved with the chomp pull-request at all, but it looks like a pretty big mess right now...

rhaschke, 2018-10-26:
@raghavendersahdev Thanks for cleaning this up. As far as I can see, there is consensus within the maintainers to proceed as proposed.

raghavendersahdev, 2018-11-15:
sorry for missing in action here, I will try to finish this up on the coming weekend by end of day Nov 18 (Sunday) . .

rhaschke, 2018-11-26:
@raghavendersahdev Do you have time to proceed on this anytime soon? Would be great to include this into the next release #1225.

raghavendersahdev, 2018-11-27:
@rhaschke unfortunately I am not certain about my time commitments right now. Haven't gotten a chance to address the yet. But if I don't open a PR addressing this by 11:59 pm on Tuesday (Nov 27) AoE (anywhere on Earth) time, I will work on Saturday (Dec 1) which will be the hard deadline for me to do this, else I do realize that I might lose the opportunity to contribute in this chomp_cleanup issue.. 

It's essentially only moving chomp related stuff/packages into a single chomp motion planner, simply copy pasting the code and including the requisite dependencies (stated in the description above) in CMakeLists.txt in an appropriate manner, right ?

rhaschke, 2018-11-27:
@raghavendersahdev Thanks for taking care of this. As stated in the main issue description, it's all about moving the code into a common folder moveit_planners/chomp and have separate packages for individual components.

rhaschke, 2018-12-08:
I will tackle this now.