moveit/moveit #1154: Moveit regression on Melodic

clalancette, 2018-10-26:
### Description

Most of Moveit is currently failing to build on the Melodic build farm.  The root of the issue seems to be that moveit_ros_planning is failing to build, which causes a cascade of failures further down the line.  It looks like moveit_ros_planning is failing to build because of a missing package.xml dependency on cmake_modules: http://build.ros.org/view/Mbin_uB64/job/Mbin_uB64__moveit_ros_planning__ubuntu_bionic_amd64__binary/16/console .  If we could get that in and released soon, that would be great as I'm planning to do a Melodic sync next week: https://discourse.ros.org/t/preparing-for-melodic-sync-2018-11-01/6608 .  Otherwise, I'll likely downgrade Moveit for that sync and we can get it fixed for the next sync cycle.  Thanks!

moriarty, 2018-10-26:
I think this is missing in `moveit_ros/planning/planning_request_adapter_plugins/CMakeLists.txt` because of #1012. I will submit a PR

moriarty, 2018-10-26:
#1012 went into the kinetic branch, and the job is failing there too.

http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__moveit_ros_planning__ubuntu_xenial_amd64__binary/88/console

rhaschke, 2018-10-26:
@davetcoleman Your bot is definitely not greeting first-time users only. Chris is a well-known friend ;-)

rhaschke, 2018-10-26:
@clalancette We already tackled this issue in #1134.
@130s promised to do a new release this weekend.

closed 2018-10-26 by rhaschke

rhaschke, 2018-10-27:
Touche! But he already file a PR before:
https://github.com/ros-planning/moveit/issues?utf8=%E2%9C%93&q=author%3Aclalancette+

clalancette, 2018-10-29:
> @clalancette We already tackled this issue in #1134.
> @130s promised to do a new release this weekend.

Perfect, thanks.  I merged the new release into Melodic, and it is building now :).