moveit/moveit #1168: Make binary releases from designated "release" branches

130s, 2018-10-29:
I suggest to add a small layer(s) of branching policies. This could be a next maintainers' meeting topic.

TLDR I suggest to secure release branches in addition to what we've been using `*-devel`.

This new (to moveit repo) branch is:
- a snapshot of a specific devel branch.
- I suggest creating a `%DISTRO%-release` or similar per each devel, e.g. `melodic-release` for `melodic-devel`.
  - Naming could cause misreading? In that case `release-%DISTRO%` is better?
- It remains frozen except right before a release is made.

Reasonings for the suggestion:
- (Branching in general) It's just a good software maintenance practice to separate branches for purpose. Ref. e.g. https://nvie.com/posts/a-successful-git-branching-model/
- This guarantees the state of the branch that goes to be released.
  - As a release maintainer I've always worried that the branch could change during release task cycle is running, as there's no mechanism to prevent merging PRs.
- Naming of *-devel convention suggests that they are meant to be `develop` branches.
- No impact for the developers by this change. No one except release maintainers should need these branches (although the branches are free to be used for any purpose).

TODO
- [ ] Decision among maintainers and/or community.
- [ ] Create release branches.

gavanderhoorn, 2018-10-29:
:+1: from me.

I've been doing this for as long as I've released software (and the repository was under my control / the release-branching strategy/policy was mine to make).

rhaschke, 2018-10-29:
Would you keep a release branch for every release made? Or would you use it only during the release process? In the latter case, I agree. In the former, I don't see the necessity to keep release branches as we do have release tags anyway.

In the latter case, I picture the release process as follows:
1. Create a new (local?) branch melodic-release at the current HEAD (or some stable state on *-devel)
2. Run pre-release tests (locally), eventually pushing some hot-fixes to this branch
3. `catkin_generate_changelog`
4. `catkin_prepare_release`, eventually pushing the new tag
5. `bloom`
6. Await success on ROS build farm
7. Merge or rebase release-branch changes into eventually progressed devel branch
8. Remove release-branch

I think your release branch could be local in this case, couldn't it?

130s, 2018-11-01:
@rhaschke Good point. After much thoughts, I agreed with your "latter" idea and the workflow except that a few steps might need small changes.

> 4. `catkin_prepare_release`, eventually pushing the new tag

I'd run `catkin_prepare_release --no-push`. This seems to allow tagging to happen locally and push to the remote later, so that release task won't be affected by the change in the remote (it may sound silly but honestly finding this option is a game changer for me regarding this discussion).

> 3. `catkin_generate_changelog`

No push to the remote at this point.

(Side-note btw. I usually edit change logs manually after auto-generating them by the command, and I admit this step is the main reason why release takes so much time for me (but I'm still happy to take on the burden if that's ok)).

With above, the following one step can be dropped.

> 1. Create a new (local?) branch melodic-release at the current HEAD (or some stable state on *-devel)

To summarize, no need for keeping release branches locally nor remotely. `git` is awesome.

@gavanderhoorn Do you have any thoughts?

I can update the maintainer's doc on moveit.ros.org once we all agree.

gavanderhoorn, 2018-11-01:
Using a dedicated release branch is quite an accepted practice in a lot of git workflows.

Releasing a project may require fixups just for the release process, but those fixes / tweaks may not need to be integrated into the main development branch. That can be done on a branch, a tag cannot be changed afterwards (it can, but it shouldn't).

You could of course just create a new tag and repeat the release process. That's fine too.

davetcoleman, 2018-11-22:
Based on our last maintainer meeting, there seems to be consensus to migrate towards using a ``master`` branch. In this case, the chance of PRs into the now-inaptly named *-devel branches will go down.

I'm not convinced we should create additional branches for the release process, this is more what tags are for.

v4hn, 2018-12-21:
> I can update the maintainer's doc on moveit.ros.org once we all agree.

@130s I believe everyone agrees that we do not need the release branches upstream, but that they are useful during releasing. Can this be closed? Did you update the documentation as you proposed?

davetcoleman, 2019-01-01:
+1 closing this

closed 2019-01-11 by v4hn