moveit/moveit #1175: Libraries broken after update

anders29, 2018-10-30:
After updating the package today, the libraries appear to be broken:

/usr/bin/ld: warning: libmoveit_trajectory_execution_manager.so.0.9.15, needed by /opt/ros/kinetic/lib/libmoveit_move_group_interface.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_planning_scene_monitor.so.0.9.15, needed by /opt/ros/kinetic/lib/libmoveit_move_group_interface.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_model.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_robot_model_loader.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_profiler.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_robot_model_loader.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_state.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_trajectory_execution_manager.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_model_loader.so.0.9.15, needed by /opt/ros/kinetic/lib/libmoveit_common_planning_interface_objects.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_kinematics_base.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_kinematics_plugin_loader.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_occupancy_map_monitor.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_transforms.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_trajectory.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_collision_detection.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_planning_scene.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so, not found (try using -rpath or -rpath-link)
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `occupancy_map_monitor::OccupancyMapMonitor::OccupancyMapMonitor(boost::shared_ptr<tf::Transformer> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double)'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::~PlanningScene()'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::clearDiffs()'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `occupancy_map_monitor::OccupancyMapMonitor::startMonitor()'
/opt/ros/kinetic/lib/libmoveit_kinematics_plugin_loader.so: undefined reference to `kinematics::KinematicsBase::DEFAULT_SEARCH_DISCRETIZATION'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::pushDiffs(std::shared_ptr<planning_scene::PlanningScene> const&)'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::setCollisionObjectUpdateCallback(boost::function<void (std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action)> const&)'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `occupancy_map_monitor::OccupancyMapMonitor::excludeShape(std::shared_ptr<shapes::Shape const> const&)'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::processCollisionObjectMsg(moveit_msgs::CollisionObject_<std::allocator<void> > const&)'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::setAttachedBodyUpdateCallback(boost::function<void (moveit::core::AttachedBody*, bool)> const&)'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::propogateRobotPadding()'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::getPlanningSceneDiffMsg(moveit_msgs::PlanningScene_<std::allocator<void> >&) const'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::usePlanningSceneMsg(moveit_msgs::PlanningScene_<std::allocator<void> > const&)'
/opt/ros/kinetic/lib/libmoveit_kinematics_plugin_loader.so: undefined reference to `typeinfo for kinematics::KinematicsBase'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::getCollisionRobotNonConst()'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `occupancy_map_monitor::OccupancyMapMonitor::~OccupancyMapMonitor()'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::decoupleParent()'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::OCTOMAP_NS[abi:cxx11]'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::getTransformsNonConst()'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::processOctomapPtr(std::shared_ptr<octomap::OcTree const> const&, Eigen::Transform<double, 3, 2, 0> const&)'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `occupancy_map_monitor::OccupancyMapMonitor::setTransformCacheCallback(boost::function<bool (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ros::Time const&, std::map<unsigned int, Eigen::Transform<double, 3, 2, 0>, std::less<unsigned int>, Eigen::aligned_allocator<std::pair<unsigned int const, Eigen::Transform<double, 3, 2, 0> > > >&)> const&)'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `robot_trajectory::RobotTrajectory::clear()'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::PlanningScene(std::shared_ptr<moveit::core::RobotModel const> const&, std::shared_ptr<collision_detection::World>)'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `robot_trajectory::RobotTrajectory::RobotTrajectory(std::shared_ptr<moveit::core::RobotModel const> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::processAttachedCollisionObjectMsg(moveit_msgs::AttachedCollisionObject_<std::allocator<void> > const&)'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::getCurrentStateNonConst()'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::isEmpty(moveit_msgs::PlanningSceneWorld_<std::allocator<void> > const&)'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::getAllowedCollisionMatrixNonConst()'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `occupancy_map_monitor::OccupancyMapMonitor::publishDebugInformation(bool)'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::processPlanningSceneWorldMsg(moveit_msgs::PlanningSceneWorld_<std::allocator<void> > const&)'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `occupancy_map_monitor::OccupancyMapMonitor::stopMonitor()'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::diff() const'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::getPlanningSceneMsg(moveit_msgs::PlanningScene_<std::allocator<void> >&) const'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `occupancy_map_monitor::OccupancyMapMonitor::forgetShape(unsigned int)'
/opt/ros/kinetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::isEmpty(moveit_msgs::RobotState_<std::allocator<void> > const&)'

IanTheEngineer, 2018-10-30:
@anders29 can you please give a bit more context for this error? Are you trying to build MoveIt, or your package that depends on MoveIt? 

Have you attempted a full update with ``` sudo apt-get update && sudo apt-get dist-upgrade```? There were several Kinetic packages recently updated, and it's possible your local workstation has only partially updated.

IanTheEngineer, 2018-10-30:
Ah, I see the source of confusion: `apt-get dist-upgrade` doesn't upgrade your Ubuntu version, it just manages conflict resolution by upgrading packages and removing conflicting packages. From the manpage:
```
NAME
       apt-get - APT package handling utility -- command-line interface
...
       dist-upgrade
           dist-upgrade in addition to performing the function of upgrade, also intelligently handles
           changing dependencies with new versions of packages; apt-get has a "smart" conflict
           resolution system, and it will attempt to upgrade the most important packages at the
           expense of less important ones if necessary. The dist-upgrade command may therefore remove
           some packages. The /etc/apt/sources.list file contains a list of locations from which to
           retrieve desired package files. See also apt_preferences(5) for a mechanism for overriding
           the general settings for individual packages.
```
I'd recommend again running ```sudo apt-get update && sudo apt-get dist-upgrade``` and reporting back if you still have issues.

anders29, 2018-10-30:
Before your last comment I ran apt-get update, sudo apt-get upgrade once again and that seems to have fixed it. I guess I ran the first update before some necessary commits were released?

Closing this.

closed 2018-10-30 by anders29

RakVish1342, 2020-07-05:
> Ah, I see the source of confusion: `apt-get dist-upgrade` doesn't upgrade your Ubuntu version, it just manages conflict resolution by upgrading packages and removing conflicting packages. From the [...]

System: Ubuntu 16.04

Just adding on; Only performing ```apt update``` without the ```upgrade/dist-upgrade``` command may cause some ```.so``` files to be missing and not found while running ```catkin_make```. As suggested by IanTheEngineer and also the official docs, use of ```dist-upgrade``` after the update step resolved this. Thanks.