moveit/moveit #1178: error: cannot convert ‘moveit::planning_interface::MoveItErrorCode’ to ‘bool’ in assignment result_ = group.plan(my_plan); and Invoking "make -j8 -l8" failed

mambawockeez, 2018-10-31:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

Im using ubuntu 16.04 to do a kinova arm project and i install kinova package, after when i catkin_make it, it should error from 96%.
[image]
[image]
[image]

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

BryceStevenWilley, 2018-10-31:
Hey mambawockeez, just a heads up that questions about building issues like this should be asked on [ROS Answers](https://answers.ros.org/questions/). If it turns out it is a bug in MoveIt, then you can make an issue here. 

This is a similar issue to https://github.com/ros-planning/moveit/issues/856, I suggest you take a look at that and at the [Migration Nodes](https://github.com/ros-planning/moveit/blob/melodic-devel/MIGRATION.md), the first bullet under ROS Kinetic has the fix that you need to apply to the Kinova library.

mambawockeez, 2018-10-31:
@BryceStevenWilley Hi, thank for the help. And im really new to tos and kinova, could you please be more specifit that what do i need to do? I read the first bullet in Migration Nodes ROS Kinetic, but still confused that what should i do next.

BryceStevenWilley, 2018-10-31:
I'd like to help, but this Github issue isn't the best place to do it.  Ask this question on ROS Answers, and post the link here, I can help you more there.

mambawockeez, 2018-10-31:
@BryceStevenWilley will do! thanks a lot!

mambawockeez, 2018-11-01:
@BryceStevenWilley https://answers.ros.org/question/307399/error-cannot-convert-moveitplanning_interfacemoveiterrorcode-to-bool-in-assignment-result_-groupplanmy_plan-and-invoking-make-j8-l8-failed/ Hi this is the link that goes to ROS answer, please look at it and help me a little bit! Many thanks!

mambawockeez, 2018-11-08:
@BryceStevenWilley Do you have time to check my new issue? it seems like i got problem with gazebo simulation. https://github.com/Kinovarobotics/kinova-ros/issues/180