moveit/moveit #1183: [indigo] SONAME not updated since 0.7.3?

130s, 2018-11-01:
### Description

Just notice on 2 of my Ubuntu 14.04 computer/container, I see `SONAME` for `libmoveit*` files are 0.7.3. Latest release is 0.7.13 (0.7.14 is waiting on shadow repo).
On 16.04 this is not the case for me.

### Criticality

Not sure how long this has been sitting there but I assume it has been since 0.7.3 release? If so not critical at all by itself (then I wonder if this means that SONAME might have not been functioning as we expect?).

### Your environment
* ROS Distro: Indigo
* OS Version: Ubuntu 14.04
* Binary build, 0.7.13 and 0.7.14 (taken from shadow repo)

### Steps to reproduce

Reproducible on Docker.
```
docker pull ros:indigo-ros-core
docker images | grep "indigo-ros-core"
docker run -it ros:indigo-ros-core
find /opt/ros/indigo/ -iname libmoveit*  (returns 0 result)
apt-get update && apt-get install ros-indigo-moveit
find /opt/ros/indigo/ -iname libmoveit*
```

### Expected behaviour

Should be suffixed by 0.7.13 e.g.

```
$ find /opt/ros/indigo/ -iname libmoveit*
/opt/ros/indigo/lib/libmoveit_kinematics_plugin_loader.so
/opt/ros/indigo/lib/libmoveit_mesh_filter.so
/opt/ros/indigo/lib/libmoveit_collision_plugin_loader.so
/opt/ros/indigo/lib/libmoveit_lazy_free_space_updater.so
/opt/ros/indigo/lib/libmoveit_srv_kinematics_plugin.so.0.7.13
:
/opt/ros/indigo/lib/libmoveit_motion_planning_rviz_plugin_core.so
/opt/ros/indigo/lib/libmoveit_collision_plugin_loader.so
/opt/ros/indigo/lib/libmoveit_lma_kinematics_plugin.so.0.7.13
/opt/ros/indigo/lib/libmoveit_collision_detection_fcl.so
/opt/ros/indigo/lib/libmoveit_srv_kinematics_plugin.so
/opt/ros/indigo/lib/libmoveit_default_planning_request_adapter_plugins.so
/opt/ros/indigo/lib/libmoveit_planning_scene_rviz_plugin.so
/opt/ros/indigo/lib/libmoveit_kinematics_plugin_loader.so
/opt/ros/indigo/lib/libmoveit_common_planning_interface_objects.so.0.7.13
/opt/ros/indigo/lib/libmoveit_warehouse.so.0.7.13
/opt/ros/indigo/lib/libmoveit_move_group_default_capabilities.so.0.7.13
/opt/ros/indigo/lib/libmoveit_collision_plugin_loader.so.0.7.13
/opt/ros/indigo/lib/libmoveit_mesh_filter.so
/opt/ros/indigo/lib/libmoveit_ros_control_interface_trajectory_plugin.so.0.7.13
:
```

### Actual behaviour

```
$ find /opt/ros/indigo/ -iname libmoveit*
/opt/ros/indigo/lib/libmoveit_kinematics_plugin_loader.so
/opt/ros/indigo/lib/libmoveit_mesh_filter.so
/opt/ros/indigo/lib/libmoveit_collision_plugin_loader.so
/opt/ros/indigo/lib/libmoveit_lazy_free_space_updater.so
/opt/ros/indigo/lib/libmoveit_srv_kinematics_plugin.so.0.7.3
:
/opt/ros/indigo/lib/libmoveit_motion_planning_rviz_plugin_core.so
/opt/ros/indigo/lib/libmoveit_collision_plugin_loader.so
/opt/ros/indigo/lib/libmoveit_lma_kinematics_plugin.so.0.7.3
/opt/ros/indigo/lib/libmoveit_collision_detection_fcl.so
/opt/ros/indigo/lib/libmoveit_srv_kinematics_plugin.so
/opt/ros/indigo/lib/libmoveit_default_planning_request_adapter_plugins.so
/opt/ros/indigo/lib/libmoveit_planning_scene_rviz_plugin.so
/opt/ros/indigo/lib/libmoveit_kinematics_plugin_loader.so
/opt/ros/indigo/lib/libmoveit_common_planning_interface_objects.so.0.7.3
/opt/ros/indigo/lib/libmoveit_warehouse.so.0.7.3
/opt/ros/indigo/lib/libmoveit_move_group_default_capabilities.so.0.7.3
/opt/ros/indigo/lib/libmoveit_collision_plugin_loader.so.0.7.3
/opt/ros/indigo/lib/libmoveit_mesh_filter.so
/opt/ros/indigo/lib/libmoveit_ros_control_interface_trajectory_plugin.so.0.7.3
:
```

Btw, using shadow repo, 0.7.14 can be installed but the result were the same. So initially I thought for 0.7.13 the number of tenth order dropped to make 0.7.3, but that's not the case.

### Backtrace or Console output

```
root@5ac9bea26442:/# apt-cache policy moveit
N: Unable to locate package moveit
root@5ac9bea26442:/# apt-cache policy ros-indigo-moveit
ros-indigo-moveit:
  Installed: 0.7.13-0trusty-20181006-132937-0800
  Candidate: 0.7.13-0trusty-20181006-132937-0800
  Version table:
 *** 0.7.13-0trusty-20181006-132937-0800 0
        500 http://packages.ros.org/ros/ubuntu/ trusty/main amd64 Packages
        100 /var/lib/dpkg/status
root@5ac9bea26442:/# apt-cache policy ros-indigo-moveit-core
ros-indigo-moveit-core:
  Installed: 0.7.13-0trusty-20180724-044429-0800
  Candidate: 0.7.13-0trusty-20180724-044429-0800
  Version table:
 *** 0.7.13-0trusty-20180724-044429-0800 0
        500 http://packages.ros.org/ros/ubuntu/ trusty/main amd64 Packages
        100 /var/lib/dpkg/status
```
shadow
```
$ apt-cache policy ros-indigo-moveit-core
ros-indigo-moveit-core:
  Installed: 0.7.14-1trusty-20181029-115419-0800
  Candidate: 0.7.14-1trusty-20181029-115419-0800
  Version table:                          
 *** 0.7.14-1trusty-20181029-115419-0800 0
        500 http://packages.ros.org/ros-shadow-fixed/ubuntu/ trusty/main amd64 Packages
        100 /var/lib/dpkg/status                             
```

rhaschke, 2018-11-01:
SONAME is hardcoded in cmake files to 0.7.3 for Indigo. Maybe, @v4hn remembers the details?

v4hn, 2018-11-05:
SONAMES have *nothing* to do with the version number of the whole framework.
Think of them as independent version numbers of the individual libraries.
We pretty much abuse this system by forcing this relationship in kinetic+, because we need it to make ABI-breaking changes easily.

There is absolutely nothing wrong with the indigo release.

closed 2018-11-05 by v4hn