moveit/moveit #1186: Moveit_msgs has no member named planing_time

abhileshborode, 2018-11-04:
### Description

I tried building the Moveit package using catkin_make, it threw a make error with the message 

/home/turtlebot/moveit_try/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp: In member function ‘void move_group::MoveGroupPickPlaceAction::executePickupCallback_PlanOnly(const PickupGoalConstPtr&, moveit_msgs::PickupResult&)’:
/home/turtlebot/moveit_try/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp:120:18: error: ‘moveit_msgs::PickupResult {aka struct moveit_msgs::PickupResult_<std::allocator<void> >}’ has no member named ‘planning_time’
       action_res.planning_time = plan->getLastPlanTime();
                  ^
/home/turtlebot/moveit_try/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp: In member function ‘void move_group::MoveGroupPickPlaceAction::executePlaceCallback_PlanOnly(const PlaceGoalConstPtr&, moveit_msgs::PlaceResult&)’:
/home/turtlebot/moveit_try/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp:160:18: error: ‘moveit_msgs::PlaceResult {aka struct moveit_msgs::PlaceResult_<std::allocator<void> >}’ has no member named ‘planning_time’
       action_res.planning_time = plan->getLastPlanTime();
                  ^
/home/turtlebot/moveit_try/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp: In member function ‘bool move_group::MoveGroupPickPlaceAction::planUsingPickPlace_Pickup(const PickupGoal&, moveit_msgs::PickupResult*, plan_execution::ExecutableMotionPlan&)’:
/home/turtlebot/moveit_try/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp:201:19: error: ‘moveit_msgs::PickupResult {aka struct moveit_msgs::PickupResult_<std::allocator<void> >}’ has no member named ‘planning_time’
       action_res->planning_time = pick_plan->getLastPlanTime();
                   ^
/home/turtlebot/moveit_try/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp: In member function ‘bool move_group::MoveGroupPickPlaceAction::planUsingPickPlace_Place(const PlaceGoal&, moveit_msgs::PlaceResult*, plan_execution::ExecutableMotionPlan&)’:
/home/turtlebot/moveit_try/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp:244:19: error: ‘moveit_msgs::PlaceResult {aka struct moveit_msgs::PlaceResult_<std::allocator<void> >}’ has no member named ‘planning_time’
       action_res->planning_time = place_plan->getLastPlanTime();
                   ^

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04
* Source 
kinetic-devel

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

RDaneelOlivav, 2018-11-21:
Having same issue here, trying to compile in a Python3 Environment and gives these errors:

`/home/user/ai_ws/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp: In member function ‘void move_group::MoveGroupPickPlaceAction::executePickupCallback_PlanOnly(const PickupGoalConstPtr&, moveit_msgs::PickupResult&)’:
/home/user/ai_ws/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp:120:18: error: ‘moveit_msgs::PickupResult {aka struct moveit_msgs::PickupResult_<std::allocator<void> >}’ has no member named ‘planning_time’
       action_res.planning_time = plan->getLastPlanTime();
                  ^
/home/user/ai_ws/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp: In member function ‘void move_group::MoveGroupPickPlaceAction::executePlaceCallback_PlanOnly(const PlaceGoalConstPtr&, moveit_msgs::PlaceResult&)’:
/home/user/ai_ws/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp:160:18: error: ‘moveit_msgs::PlaceResult {aka struct moveit_msgs::PlaceResult_<std::allocator<void> >}’ has no member named ‘planning_time’
       action_res.planning_time = plan->getLastPlanTime();
                  ^
/home/user/ai_ws/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp: In member function ‘bool move_group::MoveGroupPickPlaceAction::planUsingPickPlace_Pickup(const PickupGoal&, moveit_msgs::PickupResult*, plan_execution::ExecutableMotionPlan&)’:
/home/user/ai_ws/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp:201:19: error: ‘moveit_msgs::PickupResult {aka struct moveit_msgs::PickupResult_<std::allocator<void> >}’ has no member named ‘planning_time’
       action_res->planning_time = pick_plan->getLastPlanTime();
                   ^
/home/user/ai_ws/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp: In member function ‘bool move_group::MoveGroupPickPlaceAction::planUsingPickPlace_Place(const PlaceGoal&, moveit_msgs::PlaceResult*, plan_execution::ExecutableMotionPlan&)’:
/home/user/ai_ws/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp:244:19: error: ‘moveit_msgs::PlaceResult {aka struct moveit_msgs::PlaceResult_<std::allocator<void> >}’ has no member named ‘planning_time’
       action_res->planning_time = place_plan->getLastPlanTime();
                   ^
moveit/moveit_ros/manipulation/move_group_pick_place_capability/CMakeFiles/moveit_move_group_pick_place_capability.dir/build.make:62: recipefor target 'moveit/moveit_ros/manipulation/move_group_pick_place_capability/CMakeFiles/moveit_move_group_pick_place_capability.dir/src/pick_place_action_capability.cpp.o' failed
make[2]: *** [moveit/moveit_ros/manipulation/move_group_pick_place_capability/CMakeFiles/moveit_move_group_pick_place_capability.dir/src/pick_place_action_capability.cpp.o] Error 1
CMakeFiles/Makefile2:17839: recipe for target 'moveit/moveit_ros/manipulation/move_group_pick_place_capability/CMakeFiles/moveit_move_group_pick_place_capability.dir/all' failed
make[1]: *** [moveit/moveit_ros/manipulation/move_group_pick_place_capability/CMakeFiles/moveit_move_group_pick_place_capability.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2 -l2" failed`

rhaschke, 2018-11-21:
Looks like you are trying to compile the `melodic-devel` branch on a `Kinetic` installation.
To do so, you need to also update/build [`moveit_msgs` from melodic-devel](https://github.com/ros-planning/moveit_msgs).