moveit/moveit #1188: Does Moveit support incrementally built Octomap?

jacoblim20, 2018-11-05:
I am unable to build an octomap incrementally, by setting "octomap_frame = odom" in husky_ur5_moveit_sensor_manager.launch.xml.

### Your environment
* ROS Distro: [Indigo]
* Ubuntu 14.04
* Binary build 0.7.13-Alpha

### Steps to reproduce
I followed this [perception tutorial](http://docs.ros.org/indigo/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/perception_configuration.html)

### Expected behaviour
Incrementally built octomap when the robot is moved around (as octomap_server)

### Actual behaviour
The octomap seems to be centered on the robot and obstacles are occupying more than one location.
[image]

simonschmeisser, 2018-11-12:
So according to the tutorial the octomap frame should be fixed ie "world_frame" or similar. Your PointCloud/DepthMap should have "odom" (the tf frame of the mobile robot) set as its reference frame.

If you still don't succeed please post more of your config or an example package link.

v4hn, 2018-12-21:
No feedback, closing.

closed 2018-12-21 by v4hn

sdfryc-uts, 2019-02-12:
I am also experiencing this issue. Setting "octomap_frame = odom" inside the standalone octomap server builds a map as expected.

But setting  "octomap_frame = odom" when configuring the moveit sensor manager launch file seems to have no effect.

- ROS Distro: [kinetic] 
- Ubuntu 16.04

 @simonschmeisser could you please clarify what you mean by "PointCloud/DepthMap should have "odom" (the tf frame of the mobile robot) set as its reference frame (ie in the package header)."?

My sensor configuration is:
```

sensors:
  - sensor_plugin: occupancy_map_monitor/PointCloudOctomapUpdater
    point_cloud_topic: /sensor_d435/depth/points
    max_range: 5.0
    point_subsample: 1
    padding_offset: 0.1
    padding_scale: 1.0
    max_update_rate: 20.0
    filtered_cloud_topic: filtered_cloud
```

My sensor manager launch file configuration is:

```
<launch>

  <!-- This file makes it easy to include the settings for sensor managers -->

  <!-- Params for 3D sensors config -->
  <rosparam command="load" file="$(find schunk_moveit_config)/config/sensors_3d.yaml" />

  <!-- Params for the octomap monitor -->
  <param name="octomap_frame" type="string" value="odom" />
  <param name="octomap_resolution" type="double" value="0.05" />
  <param name="max_range" type="double" value="5.0" />

  <!-- Load the robot specific sensor manager; this sets the moveit_sensor_manager ROS parameter -->
  <arg name="moveit_sensor_manager" default="mp_700" />
  <include file="$(find schunk_moveit_config)/launch/$(arg moveit_sensor_manager)_moveit_sensor_manager.launch.xml" />

</launch>
```

sdfryc-uts, 2019-02-27:
I now understand that the planningscene frame is set to the root link in the URDF/SRDF by default. In my case, this caused the octomap to always update in the frame of my mobile robot. The solution was to add a virtual joint in my SRDF that connects the fixed frame (map) to the root link on my robot.

Having said that, it appears that within  "occupancy_map_monitor.cpp" the param octomap_frame is never set because the map frame always defaults to the root link of the SRDF / URDF.

```
if (map_frame_.empty())
    if (!nh_.getParam("octomap_frame", map_frame_))
      if (tf_)
        ROS_WARN("No target frame specified for Octomap. No transforms will be applied to received data.");
```
In my opinion the first two IF statements should be in reverse order, such that, IF a custom map_frame is not specified by the user, THEN default to the root link. I am a fairly new user, happy to be corrected. Thanks

yanliang-wang, 2021-10-14:
> I now understand that the planningscene frame is set to the root link in the URDF/SRDF by default. In my case, this caused the octomap to always update in the frame of my mobile robot. The solution was to add a virtual joint in my SRDF that connects the fixed frame (map) to the root link on my robot.
> 
> Having said that, it appears that within "occupancy_map_monitor.cpp" the param octomap_frame is never set because the map frame always defaults to the root link of the SRDF / URDF.
> 
> ```
> if (map_frame_.empty())
>     if (!nh_.getParam("octomap_frame", map_frame_))
>       if (tf_)
>         ROS_WARN("No target frame specified for Octomap. No transforms will be applied to received data.");
> ```
> 
> In my opinion the first two IF statements should be in reverse order, such that, IF no custom map_frame is specified by the user, THEN default to the root link. I am a fairly new user, happy to be corrected. Thanks

Thank you very much for your solution. I also experienced the issue and solved it by adding a virtual joint(with a `floating` type)  as you said. 

I also tried to reverse the IF statements but it didn't work. I think more modifications need to be done to solve this bug. Hope the maintainer can pay some attention to this bug. @v4hn , thank you in advance.