moveit/moveit #1195: Class vs Struct definition

S-Fichtl, 2018-11-12:
### Description

Inconsistent usage of class and struct for the same data type.
https://github.com/ros-planning/moveit/blob/melodic-devel/moveit_core/collision_detection/include/moveit/collision_detection/world.h#L77
here in line 77 `Object` is defined as a class but in the line 80 right below it is defined as a struct. This leads to issues when compiling a project that uses moveit with llvm/clang.

> In file included from /opt/ros/kinetic/include/moveit/collision_detection/collision_world.h:44:
> /opt/ros/kinetic/include/moveit/collision_detection/world.h:80:3: error: 'Object' defined as a struct here > but previously declared as a class [-Werror,-Wmismatched-tags]
>   struct Object
>   ^
> /opt/ros/kinetic/include/moveit/collision_detection/world.h:77:3: note: did you mean struct here?
>   class Object; typedef std::shared_ptr<Object> ObjectPtr; typedef std::shared_ptr<const Object> ObjectConstPtr;;;
>  ^~~~~
>  struct

### Your environment
* ROS Distro: [Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Binary build

### Steps to reproduce
I suppose to reproduce it would be enough to create a project that directly or indirectly includes `/opt/ros/kinetic/include/moveit/collision_detection/world.h` and creates an instance of type `Object` and then compile using llvm/clang

### Expected behaviour
an instance of type `Object` should be instanciated.

### Actual behaviour
a compiler error like the one above.

### Notes
I don't think it's required to actually reproduce locally as the issue seems to be a straight forward mismatch between `struct` and `class` usage. Just make both class or both struct.

jonbinney, 2018-11-12:
Interesting - seems like we could either change `Object` to a class or add a `MOVEIT_STRUCT_FORWARD` macro that handles structs appropriately.

S-Fichtl, 2018-11-13:
The above is not the only place where this happens:

> /opt/ros/kinetic/include/moveit/planning_scene/planning_scene.h:967:3: error: 'CollisionDetector' defined as a struct here but previously declared as a class [-Werror,-Wmismatched-tags]
>   struct CollisionDetector
>   ^
> /opt/ros/kinetic/include/moveit/planning_scene/planning_scene.h:964:3: note: did you mean struct here?
>   class CollisionDetector; typedef std::shared_ptr<CollisionDetector> CollisionDetectorPtr; typedef std::shared_ptr<const CollisionDetector> CollisionDetectorConstPtr;;;
>   ^~~~~
>   struct
> 2 errors generated.

Should I create a separate issue for that or is this here fine? @rhaschke ?

S-Fichtl, 2018-11-16:
@v4hn There is already PR #1196 created by @rhaschke 
He opted for the other direction and turn the structs into classes, though.

rhaschke, 2018-11-16:
Indeed, I did. But I didn't look either for it. Would be the better solution... I will change this.

PR #1196 (2018-11-19): Various fixes to build MoveIt with clang
Fixes #1195.
closed 2018-11-19 by v4hn