moveit/moveit #1206: No such file or directory

boonflies, 2018-11-19:
### Description

I tried to add a new planner by including in the file, planning_context_manager.cpp
#include <ompl/geometric/planners/rrt/newRRT.h>
but when i build, catkin build, i get the following error
Errors     << moveit_planners_ompl:make /home/dyana/catkin_ws/logs/moveit_planners_ompl/build.make.015.log
/home/dyana/catkin_ws/src/moveit/moveit_planners/ompl/ompl_interface/src/planning_context_manager.cpp:45:46: fatal error: ompl/geometric/planners/rrt/newRRT.h: No such file or directory
compilation terminated.
make[2]: *** [ompl_interface/CMakeFiles/moveit_ompl_interface.dir/src/planning_context_manager.cpp.o] Error 1
make[1]: *** [ompl_interface/CMakeFiles/moveit_ompl_interface.dir/all] Error 2
make: *** [all] Error 2
cd /home/dyana/catkin_ws/build/moveit_planners_ompl; catkin build --get-env moveit_planners_ompl | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source build
* Kinetic-devel

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

BryceStevenWilley, 2018-11-19:
How are you building OMPL with this new RRT? OMPL and catkin have historically had issues in the past, and if you still have a system installed OMPL as well as an OMPL from source, it could cause include/linking errors see https://github.com/ros-planning/moveit/issues/169 and the linked issues from that.

boonflies, 2018-11-25:
I followed the steps as in the attached text file.
[moveit_ompl_installation.txt](https://github.com/ros-planning/moveit/files/2613023/moveit_ompl_installation.txt)

BryceStevenWilley, 2018-11-25:
From the steps you followed, I think the issues is near the bottom, where you manually copy `libompl.so` to `opt/ros/kinetic/lib/k86_64-linux-gnu`. I'm not sure how the new way of overlaying system-installed OMPL (`sudo apt-get install ros-kinetic-ompl`) with catkin OMPL is supposed to work, but it shouldn`t involve manually moving libraries around. 

Notable, when trying to reproduce this, after building OMPL, catkin suggests: `Note: Workspace packages have changed, please re-source setup files to use them.` This means you should run `source ~/catkin_ws/devel/setup.bash` in between build commands when building new packages like OMPL. That might fix the problem with the OMPL library not being found.