moveit/moveit #1212: Duplicate point in concatenated trajectory.

martiniil, 2018-11-21:
Consider two trajectories traj1 and traj2, where the start state of traj2 matches the end state of traj1. 
The result of appending traj2 to traj1 has a duplicate point. Unfortunately, there is no way of sanitizing a robot_trajectory::RobotTrajectory by e.g. removing duplicate points.

Our goal is to concatenate multiple trajectories given in the form of robot_trajectory::RobotTrajectory objects: https://github.com/PilzDE/pilz_industrial_motion/tree/kinetic-devel/pilz_trajectory_generation

As we do not want to increase the overall duration, we append with time interval dt=0.0. But controllers, so far at least the ros_controllers::JointTrajectoryController require a timewise strictly increasing trajectory.

We implemented a work-around, but it would be nice, if this issue gets solved on the moveit side.
PilzDE/pilz_industrial_motion@5e37860044eb6f467bae51eea4f7f39a81488225

simonschmeisser, 2018-11-28:
This can now be closed, right?

rhaschke, 2018-11-28:
At least #1213 implements a clean solution to concatenate these solutions.
However, there is no general method available to remove duplicate points from trajectories.