moveit/moveit #1217: Attached collision objects are not collision checking

jkarimi91, 2018-11-21:
### Description
Attached collision objects are not collision checking.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Binary build
* If binary, which release version? 0.9.15

### Steps to reproduce
  1. Setup the [MoveIt! Tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/getting_started/getting_started.html). 
  2. `roslaunch panda_moveit_config demo.launch`
  3. Add `Motion Planning` to the Rviz Displays
  4. `rosrun moveit_tutorials attach_object_async.py`
  5. move the end effector, via rviz, such that the attach object is in collision with the non-attached object
  6. plan the path from the motion planning rviz plugin

[image]

### Expected behavior
Planning should fail due to collision between the attached collision object and the non-attached collision object.

### Actual behaviour
Planning succeeds.

### Backtrace or Console output
NA

rhaschke, 2018-11-30:
I was not able to reproduce the problem.

> rosrun moveit_tutorials attach_object_async.py

`moveit_tutorials` doesn't provide a python script `attach_object_async.py` (maybe tutorial maintainers can provide more insight?), so I'm not sure what you actually did. Please provide the script.
Attaching an object myself, doesn't pose any issues with collision checking during planning.

rhaschke, 2018-11-30:
Thanks! Now, I was able to reproduce your issue. I think there is an issue in the PlanningSceneInterface (or its python wrapper). Unfortunately, I don't have time today to dive into this. 

Please consider using an alternative approach to attach the object: use `scene.add_*()` to instantiate the object and `group.attach_object("tool", "panda_link8")` to attach it. 

Full (working) example: [test.py](https://github.com/ros-planning/moveit/files/2632510/test.txt)

sjhansen3, 2018-11-30:
Thank you @rhaschke we will take a look at this and get back to you if its not fixing it on our end. 

I'd be curious to know what might be causing this issue. We have been considering removing the use of the python move_group and instead publishing the ros messages ourselves. Publishing messages directly to the move_group node didn't seem to fix the issue either.

sjhansen3, 2018-11-30:
@davetcoleman - This is the issue we were discussing earlier :)

davetcoleman, 2018-12-03:
I've replicated this issue on my side as well, thanks for this easy to follow script!

To aid in debugging, I made a [quick utility tool](https://github.com/ros-planning/moveit/pull/1239) that verifies the objects are being properly added / attached: *this isn't an issue with how collision objects are added*

After further debugging I'm finding that the goal state properly prohibits collisions with the attached object and detached object. *Only the start state is ignoring attached objects*.

I have a number of guesses as to why this is, but will have to continue this investigation later.

rhaschke, 2018-12-06:
@sjhansen3 I cannot reproduce the issue anymore! Your gist example actually contains several python syntax (indentation) errors. Last time, I "fixed" it the wrong way, returning `False` in the first loop of `wait_for_state_update()`. Now, having fixed this, the code works like a charm.
Previously, I guess python was quitting to soon, to be able to actually send the PlanningScene change.

[working.txt](https://github.com/ros-planning/moveit/files/2653285/working.txt)

jkarimi91, 2018-12-06:
@rhaschke I will take a look into this and get back to you today.

v4hn, 2018-12-21:
@jkarimi91 is this still relevant?

SamuelCheong, 2019-04-10:
Hi, this is samuel.

I'm not sure if I should open a new issue but I encountered a similar situation. I do hope this thread is still open and relevant. I was attempting for my robot to hold a tool and perform various stuff, however, the attached tool keeps going into other collision objects. So far, using the rviz manually, my robot does avoid the collision objects, but using movegroup, it does not.

So I attempted to reproduce the attached object collision problem with the moveit tutorial (link as below).
http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/move_group_interface/move_group_interface_tutorial.html

[image]

The code which can be pulled from the moveit tutorial github. But I added some edits to make the end_effector attempt to go to a place near a collision object. So I only added minor changes to spawn a new attach object and designate a new position to go to. 

My whole tutorial plus my edits can be seen here:
https://github.com/SamuelCheong/testing/blob/master/move_group_interface/src/move_group_interface_tutorial.cpp

So i added the cube from line 370 onwards and attempted to go near the other collision object.
I am on ubuntu 16.04 and kinetic. Is the movegroup not suppose to avoid collision of the attached object with other collision objects? Or am I not doing it correctly? 

I was expecting the planning to fail too, which it did not.

Please go easy on me. Any help will be much appreciated, and thank you for your time.

Samuel

SamuelCheong, 2019-04-10:
Due to possible plagiarism, i have removed the github link.
The following edits are performed on move_group_interface_tutorial.cpp in https://github.com/ros-planning/moveit_tutorials/tree/kinetic-devel/doc/move_group_interface/src

> So I only added minor changes to spawn a new attach object and designate a new position to go to.
>My whole tutorial plus my edits can be seen here:
>https://github.com/SamuelCheong/testing/blob/master/move_group_interface/src/move_group_interface_tutorial.cpp

The minor changes i added as stated below:

` // Define a collision object ROS message.
  moveit_msgs::CollisionObject collision_puck;
  collision_puck.header.frame_id = move_group.getPlanningFrame();

  // The id of the object is used to identify it.
  collision_puck.id = "puck";

  // Define a box to add to the world.
  primitive.type = primitive.BOX;
  primitive.dimensions.resize(3);
  primitive.dimensions[0] = 0.08;
  primitive.dimensions[1] = 0.08;
  primitive.dimensions[2] = 0.08;

  // Define a pose for the box (specified relative to frame_id)
  geometry_msgs::Pose puck_pose;
  puck_pose.orientation.w = 1.0;

  puck_pose.position.x = 0.3;
  puck_pose.position.y = -0.0;
  puck_pose.position.z = 0.43;

  collision_puck.primitives.push_back(primitive);
  collision_puck.primitive_poses.push_back(puck_pose);
  collision_puck.operation = collision_puck.ADD;

  collision_objects.push_back(collision_puck);

  // Now, let's add the collision object into the world
  ROS_INFO_NAMED("tutorial", "Add a puck into the world");
  planning_scene_interface.addCollisionObjects(collision_objects);

  visual_tools.trigger();
  visual_tools.prompt("next step");

  // Now, let's attach the collision object to the robot.
  ROS_INFO_NAMED("tutorial", "Attach the object to the robot");
  move_group.attachObject(collision_puck.id);
  sleep(4.0);

  // Show text in RViz of status
  visual_tools.publishText(text_pose, "Object attached to robot", rvt::WHITE, rvt::XLARGE);
  visual_tools.trigger();
  visual_tools.prompt("next step");

  // Now when we plan a trajectory it will avoid the obstacle
  move_group.setStartState(*move_group.getCurrentState());
  geometry_msgs::Pose another_pose2;
  another_pose2.orientation.w = 1.0;
  another_pose2.position.x = 0.4;
  another_pose2.position.y = -0.2;
  another_pose2.position.z = 0.64;
  move_group.setPoseTarget(another_pose2);

  success = (move_group.plan(my_plan) == moveit::planning_interface::MoveItErrorCode::SUCCESS);
  ROS_INFO_NAMED("tutorial", "Visualizing plan 5 (pose goal move around cuboid) %s", success ? "" : "FAILED");

  // Visualize the plan in RViz
  visual_tools.deleteAllMarkers();
  visual_tools.publishText(text_pose, "Obstacle Goal", rvt::WHITE, rvt::XLARGE);
  visual_tools.publishTrajectoryLine(my_plan.trajectory_, joint_model_group);
  visual_tools.trigger();

  if(success)
  {
    move_group.execute(my_plan);
}

  visual_tools.prompt("next step");
`

sjhansen3, 2019-04-10:
This issue definitely still exists for us @v4hn. I think what we need is a piece of minimal code which reproduces the error. Let me see if I can create a minimal script which reproduces the error with the demo code.

rhaschke, 2019-04-10:
@sjhansen3 Having a minimal code snippet would be indeed helpful :wink:

sjhansen3, 2019-04-29:
I spent a few hours tonight trying to narrow in on the issue. It seems like the attached objects arn't updating properly - Here is atleast one scenario which causes it.

Steps to reproduce:

1.    Setup the MoveIt! Tutorial.
2.    `roslaunch panda_moveit_config demo.launch`

Run the following code as a node. It prints out what it expects as outputs and why.

```python
#!/usr/bin/env python

import geometry_msgs
import rospy
import sys
import actionlib

from std_msgs.msg import Header
from geometry_msgs.msg import Pose, Vector3
from shape_msgs.msg import SolidPrimitive
from moveit_msgs.msg import AttachedCollisionObject, CollisionObject
from moveit_msgs.msg import TrajectoryConstraints, Constraints, JointConstraint, RobotState, WorkspaceParameters, MotionPlanRequest
from moveit_msgs.msg import MoveGroupActionGoal, MoveGroupAction, PlanningOptions, PlanningScene, ExecuteTrajectoryAction, PlanningSceneComponents
from moveit_msgs.srv import GetPlanningScene

def form_attach_collision_object_msg(link_name, object_name, size, pose):
    obj = AttachedCollisionObject()
    # The CollisionObject will be attached with a fixed joint to this link
    obj.link_name = link_name

    #This contains the actual shapes and poses for the CollisionObject
    #to be attached to the link
    col_obj = CollisionObject()
    col_obj.id = object_name
    col_obj.header.frame_id = link_name
    col_obj.header.stamp = rospy.Time.now()
    sp = SolidPrimitive()
    sp.type = sp.BOX
    sp.dimensions = [0.0]*3
    sp.dimensions[0] = size.x
    sp.dimensions[1] = size.y
    sp.dimensions[2] = size.z
    col_obj.primitives = [sp]
    col_obj.primitive_poses = [pose]

    # Adds the object to the planning scene. If the object previously existed, it is replaced.
    col_obj.operation = col_obj.ADD

    obj.object = col_obj

    # The weight of the attached object, if known
    obj.weight = 0.0
    return obj

def form_remove_all_attached_msg(link_name):
    obj = AttachedCollisionObject()
    # The CollisionObject will be attached with a fixed joint to this link
    obj.link_name = link_name

    col_obj = CollisionObject()

    #If action is remove and no object.id is set, all objects
    #attached to the link indicated by link_name will be removed
    col_obj.operation = col_obj.REMOVE
    obj.object = col_obj

    return obj

def form_freespace_action_goal(goal_config):
    # Form the motion planning request
[... 72 lines cut ...]
    ###
    # Plan and execute a path to a location which is collision free, this should succeed and does on the first run
    ###
    freespace_goal = form_freespace_action_goal([-0.864, 1.60, 1.04, -0.428, 0.670, 1.36, 1.253])
    move_group_client.send_goal(freespace_goal)
    move_group_client.wait_for_result()
    print('move_group result error code (Should succeed, nothing to collide with) {}'.format(move_group_client.get_result().error_code))

    ###
    # Detach all tools, (this step I believe is failing)
    ###
    attach_msg = form_remove_all_attached_msg(link_name='panda_leftfinger')
    attach_object_pub.publish(attach_msg)

    ###
    # Attach tool with new name, keep the size the same, this step should succeed but hasn't been
    ###
    attach_msg = form_attach_collision_object_msg(link_name='panda_leftfinger', object_name='some_small_tool', size=tool_size, pose=tool_pose)
    attach_object_pub.publish(attach_msg)

    freespace_goal = form_freespace_action_goal([-0.864, 1.60, 1.04, -0.428, 0.670, 1.36, 1.253])
    move_group_client.send_goal(freespace_goal)
    move_group_client.wait_for_result()
    print('move_group result error code (Should succeed, tool was added back except under a new name) {}'.format(move_group_client.get_result().error_code))
```

mvieth, 2019-05-30:
@sjhansen3 I just tried your example and am not sure what the problem is. The arm moves with some_tool attached, then detaches some_tool but does not remove it from the planning scene, then creates and attaches some_small_tool, and finally the new movement fails because the attached some_small_tool collides with the unattached some_tool - this behaviour is correct, no? Or is there only a problem under kinetic on ubuntu 16.04?

MiroslavKohut, 2019-11-19:
Please is this problem still relevant ??  I am also observing problems with attached parts in my application.

rhaschke, 2019-11-19:
@MiroslavKohut, I haven't observed a similar issue for a while. Can you please provide a minimal example (optimally using the tutorial's panda_moveit_config) to reproduce your issue? Thanks.

truhoang-ubtrobot, 2019-12-04:
@MiroslavKohut 
I encountered this issue when trying to attach an object to the gripper through MoveGroupInterface and PlanningSceneInterface and ran across this thread [answers.ros.org](https://answers.ros.org/question/257795/collision-checking-between-collision-object-and-object-attached-to-the-robot/). I switched to using "apply_planning_scene" service to manually update the allowed_collision_matrix. Here's a snippet of my solution: 

```
collision_detection::AllowedCollisionMatrix acm;
acm.setEntry(target_id, gripper_links, true);
```
where `target_id` is `std::string` and `gripper_links` is `std::vector<std::string>`.

quanvuong, 2020-05-06:
Hi everyone,

I was wondering if this is fixed. I still encounter this issue. I am using Ubuntu 18.04 with moveit installed using the standard apt-get command.

I have provided a minimal example to reproduce the problem (with a video!) in the link below

https://github.com/quanvuong/collision_object_attached_object_moveit_issue

Thank you so much!

@rhaschke  @v4hn

v4hn, 2020-05-06:
Hello @quanvuong , thank you for this beautiful minimal example!
It is great to see easy to follow instructions like that.
I can even reproduce your example with MoveIt's master branch.

However, this is no case of ignored collisions with the attached object.
If you enable the "Motion Planning - Slider" panel in RViz and look at the individual waypoints of the infeasible trajectories, you will see that not a single waypoint of the trajectory is in collision.
The problem you experience is due to overly coarse collision checks in OMPL planning.
If you add `longest_valid_segment_fraction: 0.001` to the `panda_arm:` section in `panda_moveit_config/config/ompl_planning.yaml` to change the default of `0.01`, you require 10 times more collision checks during planning and your example will find no valid solution anymore.

Depending on how small/big your collision objects are, it can make sense to change this value,
but if you set it too small, your planning will simply spent all its time running collision checks..

felixvd, 2020-05-07:
@v4hn Can you add if there is another way to make the planning safer without increasing the computational load too much, e.g. by adding padding somewhere?

I know that continuous collision checking should be added in the future, but maybe you have another suggestion for users in the meantime.

v4hn, 2020-05-07:
For such infeasible cases continuous collision checking (that would involve piecewise-linear approximations of the real arm motion and thus is also not fully safe) would actually help, yes.
I doubt it's computationally much more lightweight though.

You should pick your `longest_valid_segment_fraction` in such a way that your end-effector cannot fully pass through your objects in a single step.
Of course this depends on the scale of your manipulation actions and the size of the joint space of the manipulator.
For a [pi;-pi]-limited UR5 configuration for example, my lab currently uses a value of `0.005` for standard tabletop operations with a rather big robotiq 3-finger gripper.
If you want to manipulate chess pieces though, setting it to `0.001` is probably more reasonable.
If you handle pipets and vials, you might want to consider even finer resolution.

Of course you are free to pad your objects to keep a safe margin away from the real physical collision.

quanvuong, 2020-05-07:
Hi!

Thank you for the swift reply. I didn't even know I can set the longest valid segment in moveit! Thank for letting me know about it.

Quick question to @v4hn , when you visualize the planned trajectory using the motion planning slider, do you see the object attached to the robot hand move with the robot?

I am asking because unfortunately, setting longest_valid_segment_fraction does not solve the issue on my side. I set the value of longest_valid_segment_fraction to 0.01, but the object attached to the robot hand still go through a collision object in the world in the executed plan. 

I think that collisions between the object attached to the robot hand and a collision object in the world are being ignored during motion planning. This is because when I view the planned trajectory using the motion planning slider, I only see visualization of the motion of the robot arm, and not the object attached to the robot hand. What is weird is that, after the planned trajectory is executed, the object attached to the robot hand still follows the planned motion.

Please find another video illustration below. 

https://vimeo.com/416096144

You can see the behavior I described above from the 1 minute 15 seconds mark onwards. If the video is pixelated (vimeo takes some time to post the HD video ..., unfortunate I can not control how long they take), you can download the video from here

https://github.com/quanvuong/collision_object_attached_object_moveit_issue/blob/master/Changing%20longest_valid_segment_fraction%20does%20not%20alleviate%20the%20problem.webm

I'm using melodic installed using apt-get. Maybe this issue has already been fixed in the master branch?

Thank you!

v4hn, 2020-05-08:
To be honest I'm rather confused now.
Yesterday this fixed the issue for me (I believe you meant to write `0.0**0**1` above?).

Today, however, I can reproduce it... Additionally we added attached objects to the trajectory/trail visualization last year, but it is indeed not part of the trajectory visualization...

I'll have a closer look and report back.

v4hn, 2020-05-08:
Ok, it works for me again. :sunglasses: 
The attached objects are indeed part of the trajectory visualization, as you can see here. I enabled the trail visualization so you can see the intermediate states:

[image]

As long as the object is not part of your result trajectories, it indeed simply means that the object was not attached in the `PlanningScene` of the `move_group` node, that does the actual planning.

In your video I noticed you attached the object in RViz, but you also have to press "Publish Scene" afterwards to propagate that change to `move_group`.

Your example code uses the `MGI::attachObject` method. I wasn't even aware of that function anymore and we should actually remove it in my opinion...
It uses a similar mechanism (publishing an object message) as the RViz GUI but runs asynchronously and you can't get feedback on whether it was successful.

To use synchronous updates instead, I changed
```
    move_group.attachObject(mesh_collision_object.id);
```
to
```
    moveit_msgs::AttachedCollisionObject aco;
    aco.object.id = mesh_collision_object.id;
    aco.link_name = "panda_hand";
    planning_scene_interface.applyAttachedCollisionObject(aco);
```
and it works reliable again.

quanvuong, 2020-05-08:
Thank you so much for your help @v4hn . Unfortunately it still doesn't work on my end : (.

I narrowed down the issues further and I can reproduce the issue without having to run any custom code, only using the panda_moveit_config package and a .stl file for the mesh.

I have also tried building moveit from source and the issue still persists.

In the video below, from the mark 1:00 onwards, you can see that the object does not appear in the visualization of the planned trajectory. This is even when ```rosrun moveit_ros_planning moveit_print_planning_scene_info``` reports the object is attached to the robot. This likely indicates that the attached object is not taken into consideration by the planner. When the planned trajectory is executed, then we can see the attached object follows the motion of the arm.

https://vimeo.com/416494374

In the video above, the field Planning Scene topic in the MotionPlanning plugin in Rviz is set to ```/planning_scene```. When I change the value to ```/move_group/monitored_planning_scene```, the attached object still does not show up in the planned path. You can see the behavior in the video below:

https://vimeo.com/416494222

Both video was recorded using moveit built from source using the master branch. I checked this using the command ```printenv | grep ROS``` at the beginning of the video.

I really do appreciate any help on this matter : ).

s-trinh, 2020-05-15:
I have some issues with collision checking when an object is attached to the end-effector.
See the [following video](https://imgur.com/KUqKdH7). What is done:
- add collision objects using `planning_scene_interface.applyCollisionObjects(collision_objects)`
- attach an object using `planning_scene_interface.applyAttachedCollisionObject(aco);`
- use `move_group.setPoseTarget(target_pose);` and `move_group.move();` to move to a pose where the attached object is in collision
- move to another pose where the attached object is also in collision (see the displayed frame)

MoveIt returns a successful trajectory with a final pose where the attached object is in collision. But then planning from a starting pose in collision to a final pose in collision gives an error:
```
[ INFO] [1589577449.921560775, 15.320000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1589577449.921592729, 15.320000000]: Start state appears to be in collision with respect to group manipulator
[ INFO] [1589577449.925822532, 15.320000000]: Found a valid state near the start state at distance 1.125985 after 4 attempts
[ INFO] [1589577449.954324425, 15.352000000]: Planner configuration 'manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1589577449.971478474, 15.368000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ERROR] [1589577454.958108132, 20.368000000]: RRTConnect: Unable to sample any valid states for goal tree
[ INFO] [1589577454.958139149, 20.368000000]: RRTConnect: Created 1 states (1 start + 0 goal)
[ INFO] [1589577454.958150768, 20.368000000]: No solution found after 5.003736 seconds
[ INFO] [1589577454.965323323, 20.376000000]: Unable to solve the planning problem
[ INFO] [1589577454.965651007, 20.376000000]: ABORTED: No motion plan found. No execution attempted.
```

On [this video](https://imgur.com/L085dZx), it is the same thing except the second target pose is not in collision. This time MoveIt is successful. But moving to the first target pose is not possible:
```
[ INFO] [1589580296.414977417, 21.904000000]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1589580296.415216733, 21.904000000]: Planning attempt 1 of at most 1
[ INFO] [1589580296.416144795, 21.904000000]: Planner configuration 'manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1589580296.416581938, 21.904000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ERROR] [1589580306.423471254, 31.896000000]: RRTConnect: Unable to sample any valid states for goal tree
[ INFO] [1589580306.423500273, 31.896000000]: RRTConnect: Created 1 states (1 start + 0 goal)
[ INFO] [1589580306.423513074, 31.896000000]: No solution found after 10.007092 seconds
[ INFO] [1589580306.448152158, 31.920000000]: Unable to solve the planning problem
[ INFO] [1589580306.448462489, 31.920000000]: ABORTED: No motion plan found. No execution attempted.
```

Finally, on [this video](https://imgur.com/L13cgO6) collision checking works since MoveIt is not successful when planning a target pose where some robot links are in collision.

---

I am using ROS Kinetic on Ubuntu 16.04 with the following parameters:
```
* /move_group/planner_configs/RRTConnect/longest_valid_segment_fraction: 0.005
 * /move_group/planner_configs/RRTConnect/maximum_waypoint_distance: 0.01
 * /move_group/planner_configs/RRTConnect/range: 0.0
 * /move_group/planner_configs/RRTConnect/type: geometric::RRTCon...
```

s-trinh, 2020-05-17:
I have replaced some of my `stl` meshes by box and cylinder `shape_msgs/SolidPrimitive` primitives. This is to avoid potential issues with meshes being too complex or face normal issue.
The issue remains as it can be seen in [this video](https://imgur.com/L8ZBH1U).

---

Additional information:

I am using regular MoveIt code. The test program starts with:
```
    ros::init(argc, argv, "moveit_test");
    ros::AsyncSpinner spinner(1);
    spinner.start();

    // MoveIt
    const std::string planning_group = "manipulator";

    // Move group
    moveit::planning_interface::MoveGroupInterface move_group("manipulator");
    // Planning scene
    moveit::planning_interface::PlanningSceneInterface planning_scene_interface;
```

I am using `move_group.getPlanningFrame()` for `object_collision.header.frame_id = frame_id;`

Parameters for `longest_valid_segment_fraction` are done in `ompl_planning.yaml`.

Launch file looks like this:
```
  <param name="use_sim_time" value="true"/>

  <!-- By default, we do not start a database (it can be large) -->
  <arg name="db" default="false" />
  <!-- Allow user to specify database location -->
  <arg name="db_path" default="$(find robot_moveit)/default_warehouse_mongo_db" />

  <!-- By default, we are not in debug mode -->
  <arg name="debug" default="false" />

  <!-- Load the URDF, SRDF and other .yaml configuration files on the param server -->
  <include file="$(find robot_moveit)/launch/planning_context.launch">
    <arg name="load_robot_description" value="true"/>
  </include>

  <!-- Given the published joint states, publish tf for the robot links -->
  <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" respawn="true" output="screen" />

  <!-- Run the main MoveIt! executable without trajectory execution (we do not have controllers configured by default) -->
  <include file="$(find robot_moveit)/launch/move_group.launch">
    <arg name="allow_trajectory_execution" value="true"/>
    <arg name="fake_execution" value="false"/>
    <arg name="allowed_start_tolerance" value="0"/>
    <arg name="info" value="true"/>
    <arg name="debug" value="$(arg debug)"/>
  </include>

  <node name="rviz" pkg="rviz" type="rviz" respawn="false"
    args="-d $(find <>)/launch/rviz_MoveIt.rviz"
    output="screen">
    <rosparam command="load" file="$(find robot_moveit)/config/kinematics.yaml"/>
  </node>

  <param name="move_group/trajectory_execution/allowed_execution_duration_scaling" value="40.0" />
  <param name="move_group/trajectory_execution/execution_duration_monitoring" value="true" />

  <node name="moveit_test_collision"
        pkg="moveit_test_collision"
        type="moveit_test_collision"
        respawn="false" output="screen"
  >
  </node>
```

---

This is with ROS Kinetic. Also, for the documentation of [longest_valid_segment_fraction](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/ompl_interface/ompl_interface_tutorial.html) and `maximum_waypoint_distance` maybe `robot joint space` should be added because I found `robot’s state space` confusing.

---

I am kind of lost here :(.

s-trinh, 2020-05-26:
I have built MoveIt from source.
**Any advice** to debug the issue with for instance some `printf` in adequate functions to check that collision checking is correctly or not performed with attached object?

---

The issue is probably tricky, for instance there was this [bug](https://github.com/ros-planning/moveit/issues/136) with collision issue with attached links without geometry. In my case, I have at least two links without mesh between EE frame of the robot and tip of the tool object.

Root cause of my issue can be something tricky like this.

gapbridger, 2020-08-18:
hi I also experienced the same problem in my application where I attached an object to the robot's end effector and during the retrieving motion, the planner simply ignores the attached object which caused the result motion to collide with the bin. would you please help to continue investigating the issue, we would like to know the possible options to solve the problem. thank you very much. 
@davetcoleman @v4hn