moveit/moveit #1219: plot benchmarking results

boonflies, 2018-11-22:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Kinetic]
* OS Version:  16.04
* Source build?
* kinetic-devel

### Steps to reproduce
I ran moveit benchmarking and  processed the log with,

**rosrun moveit_ros_benchmarks moveit_benchmark_statistics.py /tmp/moveit_benchmarks/TrialQuery1_ar_drone_state_1_Query1_dyana-Inspiron-3542_2018-11-22T13:23:56.219009.log**

when I tried to create a pdf of plot, I get the output as shown

**python -p benchmark.db moveit_benchmark_statistics.py /tmp/moveit_benchmarks/TrialQuery1_ar_drone_state_1_Query1_dyana-Inspiron-3542_2018-11-22T13:23:56.219009.log**
Unknown option: -p
usage: python [option] ... [-c cmd | -m mod | file | -] [arg] ...
Try `python -h' for more information.

I also tried without the -p option,
**python benchmark.db moveit_benchmark_statistics.py /tmp/moveit_benchmarks/TrialQuery1_ar_drone_state_1_Query1_dyana-Inspiron-3542_2018-11-22T13:23:56.219009.log**

then i get the following printed,
  File "benchmark.db", line 1
SyntaxError: Non-ASCII character '\xff' in file benchmark.db on line 2, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details

Does  <plot_filename> in the command,
**python -p <plot_filename> moveit_benchmark_statistics.py <path_of_logfile>**
refer to benchmark.db?

And how do I successfully generate pdf plot?
### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

BryceStevenWilley, 2019-03-29:
The documentation here seems to be wrong. You're correct: python doesn't take a `-p` flag, but [moveit_benchmark_statistics.py](https://github.com/ros-planning/moveit/blob/822a991e0ea4f713fe672e3e337d7f0702bcc224/moveit_ros/benchmarks/scripts/moveit_benchmark_statistics.py#L537) does. I think the documentation should read.
```
python moveit_benchmark_statistics.py -p <plot_filename> <path_of_logfile>
```
I'll try to open a PR to the docs soon to fix this.

rhaschke, 2019-05-19:
Closed via https://github.com/ros-planning/moveit_tutorials/pull/309.

closed 2019-05-19 by rhaschke