moveit/moveit #1220: No octomap in Perception Pipeline Tutorial

mzwang828, 2018-11-22:
### Description

I try to use use a kinect for perception when I'm controlling my own arm, but it didn't work. I then come back to review the perception pipeline tutorial, when I run the _obstacle_avoidance_demo.launch_, I found there was no octomap shown in RViz either. 

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Binary build
* kinetic

### Steps to reproduce
I just ran the command 

> roslaunch moveit_tutorials obstacle_avoidance_demo.launch

### Expected behaviour
[image]

### Actual behaviour
[image]

and in the terminal, it shows this after you can start planning now.

> You can start planning now!
> 
> [ INFO] [1542894463.909405059]: Loading robot model 'panda'...
> [ INFO] [1542894463.943537436]: Loading robot model 'panda'...
> [ INFO] [1542894464.001528310]: Starting scene monitor
> [ INFO] [1542894464.004109747]: Listening to '/move_group/monitored_planning_scene'

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

v4hn, 2018-12-10:
@mzwang828 Thanks for the report.

@Ridhwanluthra could you have a look please?

zhubeilife, 2019-06-20:
Hi, do we have any update for this issue? I encounter the same problem.

Ridhwanluthra, 2019-07-07:
Sorry for the delay on this issue.
@mzwang828 I am unable to access your gist. I know it's probably due to the delay from my side
@zhubeilife could you share your console output, I might have a better idea then

bob-ROS, 2019-08-01:
Same problem on melodic, ubuntu 18.04. Am I supposed to edit the panda default configs or not? I don't understand the tutorial. Currently having an unedited panda_moveit_config. No octomap can be seen. My console output from `roslaunch moveit_tutorials obstacle_avoidance_demo.launc`:

SUMMARY
========

PARAMETERS
 * /joint_state_publisher/use_gui: False
 * /move_group/allow_trajectory_execution: True
 * /move_group/controller_list: [{'joints': ['pan...
 * /move_group/hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/initial: [{'pose': 'ready'...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_fake_contr...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_frame: camera_rgb_optica...
 * /move_group/octomap_resolution: 0.05
 * /move_group/panda_arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/panda_arm_hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/planner_configs/BFMTkConfigDefault/balanced: 0
 * /move_group/planner_configs/BFMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/BFMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/BFMTkConfigDefault/heuristics: 1
 * /move_group/planner_configs/BFMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/BFMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/BFMTkConfigDefault/optimality: 1
 * /move_group/planner_configs/BFMTkConfigDefault/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMTkConfigDefault/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/BiESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiESTkConfigDefault/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRTkConfigDefault/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/init_temperature: 100
 * /move_group/planner_configs/BiTRRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/type: geometric::BiTRRT
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/FMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/FMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/FMTkConfigDefault/heuristics: 0
 * /move_group/planner_configs/FMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/FMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/FMTkConfigDefault/radius_multiplier: 1.1
 * /move_group/planner_configs/FMTkConfigDefault/type: geometric::FMT
 * /move_group/planner_configs/KPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/KPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/KPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/LBKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/LBKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /move_group/planner_configs/LBTRRTkConfigDefault/epsilon: 0.4
 * /move_group/planner_configs/LBTRRTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/LBTRRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/LBTRRTkConfigDefault/type: geometric::LBTRRT
 * /move_group/planner_configs/LazyPRMkConfigDefault/range: 0.0
 * /move_group/planner_configs/LazyPRMkConfigDefault/type: geometric::LazyPRM
 * /move_group/planner_configs/LazyPRMstarkConfigDefault/type: geometric::LazyPR...
[... 189 lines cut ...]
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1564645594.736043003]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1564645594.736071259]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1564645594.830278291]: Loading robot model 'panda'...
[ INFO] [1564645595.246856208]: Loading robot model 'panda'...
[ WARN] [1564645595.306650030]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/rviz_SEVST_L_CRC8296_31261_4579711252030449593/panda_arm/kinematics_solver_attempts' from your configuration.
[ INFO] [1564645595.373002651]: Starting planning scene monitor
[ INFO] [1564645595.375990494]: Listening to '/move_group/monitored_planning_scene'
`

jtflowers, 2019-11-25:
Has anyone found a solution to this?

Jackwitt77, 2019-11-25:
Has anyone found a solution to this?

carismoses, 2020-04-02:
Having the same issue, would love some help!

Jackwitt77, 2020-04-02:
I actually solved this issue on my end by making my obstacle_avoidance_demo.launch file look like this: `<launch>
  <include file="$(find panda_moveit_config)/launch/demo.launch" />

  
  <node pkg="moveit_tutorials" type="bag_publisher_maintain_time" name="point_clouds" />

  
  
  <node pkg="tf2_ros" type="static_transform_publisher" name="to_temp_link" args="0 0.4 -0.6 0 0 0  temp_link panda_link0" />
  <node pkg="tf2_ros" type="static_transform_publisher" name="to_panda_base" args="0 0 0 0 0.2 1.92 camera_rgb_optical_frame temp_link" />

</launch>`

The "if needed" text was deceptive for me because its absolutely needed in my case. Adding the static transform publisher fixed it in my case.

carismoses, 2020-04-02:
Unfortunately that's already what my launch file looks like. The only warnings I'm getting are these, but I don't think they're causing the issue.

```
[ WARN] [1585866760.356130758]: Unable to update multi-DOF joint 'virtual_joint': Failure to lookup transform between 'world' and 'panda_link0' with TF exception: Could not find a connection between 'world' and 'panda_link0' because they are not part of the same tree.Tf has two or more unconnected trees.
[ WARN] [1585866766.647410558]: Unable to transform object from frame 'temp_link' to planning frame 'world' (Could not find a connection between 'world' and 'temp_link' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1585866766.647474331]: Unable to transform object from frame 'camera_rgb_optical_frame' to planning frame 'world' (Could not find a connection between 'world' and 'camera_rgb_optical_frame' because they are not part of the same tree.Tf has two or more unconnected trees.)
```

In the demo.launch file (from the tutorials) there is a static tf being published between the panda_link0 and world frame which appears to be broken, but I'm not sure why that would matter because nothing in the perception pipeline setup requires a world frame.

askkvn, 2020-04-24:
Do we have any solution or workaround on this? I am also facing the same issue. I have followed [this tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/perception_pipeline/perception_pipeline_tutorial.html).

**Log error:**
[ INFO] [1587723403.283003521]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1587723403.283022538]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1587723405.967994524]: Loading robot model 'panda'...
[ INFO] [1587723406.103030371]: Loading robot model 'panda'...
[ INFO] [1587723406.156926076]: Loading robot model 'panda'...
[ERROR] [1587723406.162422837]: Group 'hand' is not a chain
[ERROR] [1587723406.162999473]: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'hand'
[ERROR] [1587723406.164014084]: Kinematics solver could not be instantiated for joint group hand.

I checked topic `/move_group/filtered_cloud` but nothing published on this.

**I am using:**
Ubuntu 18.04 LTS
ROS Kinect

askkvn, 2020-04-28:
By adding static transform for `world` to `panda_link0` and `temp_link` to `world` solve octomap problem for me.  

Attaching modified `obstacle_avoidance_demo.launch`  file
```
<launch>
  <include file="$(find panda_moveit_config)/launch/demo.launch" />

  <!-- Play the rosbag that contains the pointcloud data -->
  <node pkg="moveit_tutorials" type="bag_publisher_maintain_time" name="point_clouds" />

  <!-- If needed, broadcast static tf for robot root -->

  <!-- Adding two static transforms solve octomap problem -->
  <node pkg="tf2_ros" type="static_transform_publisher" name="kinect_broadcaster" args="0 0 0 0 0 0 1 world panda_link0" />
  <node pkg="tf2_ros" type="static_transform_publisher" name="to_temp_link" args="0 0.4 -0.6 0 0 0  temp_link world" />
  
  <!-- <node pkg="tf2_ros" type="static_transform_publisher" name="to_temp_link" args="0 0.4 -0.6 0 0 0  temp_link panda_link0" /> -->
  <node pkg="tf2_ros" type="static_transform_publisher" name="to_panda_base" args="0 0 0 0 0.2 1.92 camera_rgb_optical_frame temp_link" />

</launch>
```

Philopatearashraf, 2020-05-01:
Currently I am working on a dual arm panda robots and I am trying to make this tutorial and I tried the above launch file but that what I got. any help ?

[image]

JunWeston, 2020-05-18:
please modify as bellow in obstacle_avoidance_demo.launch

  <node pkg="tf2_ros" type="static_transform_publisher" name="to_temp_link" args="0 0.4 -0.6 0 0 0 world temp_link" />
  <node pkg="tf2_ros" type="static_transform_publisher" name="to_panda_base" args="0 0 0 0 0.2 1.92 camera_rgb_optical_frame world" />

JunWeston, 2020-05-18:
please modify obstacle_avoidance_demo.launch as bellow:
<node pkg="tf2_ros" type="static_transform_publisher" name="to_temp_link" args="0 0.4 -0.6 0 0 0 world temp_link" />
  <node pkg="tf2_ros" type="static_transform_publisher" name="to_panda_base" args="0 0 0 0 0.2 1.92 camera_rgb_optical_frame world" />

130s, 2023-06-28:
FYI, for `moveit2_tutorial` (tutorial suite for moveit2, not 1), I'm porting the perception tutorial to ros2-moveit2 and successfully enabled octomap. Right now it's WIP https://github.com/ros-planning/moveit2_tutorials/pull/700