moveit/moveit #1225: 201811 release

130s, 2018-11-25:
Suggested at https://github.com/ros-planning/moveit/pull/1166#issuecomment-441137603

@rhaschke thanks for taking this task!
Can we always communicate about release in a explicitly designated ticket so that we can avoid duplicate release work?

rhaschke, 2018-11-26:
I'm waiting for the sync of [tf2_eigen on Nov, 27th](https://discourse.ros.org/t/preparing-for-melodic-sync-2018-11-13/6832/3). Which open issues / PRs (other than #1096) should be merged?
[ ] #1104
[ ] #1146
[ ] #1198

v4hn, 2018-11-26:
> Can we always communicate about release in a explicitly designated ticket

Yes please! :confused:

rhaschke, 2018-11-28:
As @davetcoleman already started the release-cycle with rviz-visual-tools, the ROS build farm is breaking:
http://build.ros.org/view/Mbin_uB64/job/Mbin_uB64__moveit_visual_tools__ubuntu_bionic_amd64__binary

For this reason, I suggest to continue with the release process. This way, we have enough soaking time before the next sync. I think, it's better to test those changes in a larger community (of shadow-fixed users) instead of restricting to "from-source builders".
@v4hn @130s @davetcoleman Opinions?

davetcoleman, 2018-11-28:
Seems a little rushed to me to make a release since we just did one...

rhaschke, 2018-11-29:
> Seems a little rushed to me to make a release since we just did one.

This release comprises many API changes and we should release them as early as possible - before many people have adopted Melodic in a production environment and then would face API changes in the middle of a stable release. I think, we agreed that big API changes like these should be merged (and released) asap before freezing API again (by the end of the year?)

2nd argument is, that the MoveIt jobs on ROS build farm are broken right now:
- dev jobs are broken, because they are missing binary releases comprising the required API changes
- bin jobs are broken, because you already triggered the release process, release rviz_visual_tools

3rd argument: The next sync will happen in about two weeks, probably with a freeze a week before again. Hence, waiting for another week, we again risk to run into the freezing phase, which makes it a little harder to merge release requests (needs extra communication).

4th argument: Releasing to shadow-fixed will reach a larger community for testing than relying on from-source-builds.

Finally, I am using my changes in production since several weeks.

davetcoleman, 2018-12-03:
These are good arguments, sounds good to me.

> bin jobs are broken, because @davetcoleman already triggered the release process by releasing rviz_visual_tools.

Notably, you are the one who [asked me to make that release](https://github.com/PickNikRobotics/rviz_visual_tools/pull/105#issuecomment-441560202) :-)

rhaschke, 2018-12-08:
I would like to the release on the weekend. Any objections?

130s, 2018-12-09:
No objections. Just noticed there are still a few open items in the list https://github.com/ros-planning/moveit/issues/1225#issue-384066739

rhaschke, 2018-12-09:
Please feel free to review them ;-)
If I don't get reviews, I will merge them anyway.

rhaschke, 2018-12-10:
Release build was successful: [Melodic build status](http://repositories.ros.org/status_page/ros_melodic_default.html).
However, as pointed out in https://github.com/ros-planning/moveit/pull/1251#issuecomment-445873674, removing `moveit_experimental` from the set of released packages, doesn't automatically uninstalls this package. Thus, when updating to 0.10.6, we get a clash:
`E: /tmp/apt-dpkg-install-i6AXUC/02-ros-melodic-moveit-core_0.10.6-0bionic.20181210.112240_amd64.deb: trying to overwrite '/opt/ros/melodic/include/moveit/collision_distance_field/collision_common_distance_field.h', which is also in package ros-melodic-moveit-experimental 0.10.5-0bionic.20181117.194840`

@clalancette Is there a way to automatically uninstall removed packages?

rhaschke, 2018-12-10:
OK, I will try this approach. Looks like the easiest one.

clalancette, 2018-12-10:
Yeah, I agree with @v4hn that we should release moveit_experimental as an empty package, that way upgrades will be smooth for users.

rhaschke, 2018-12-11:
Although, upgrading worked in principle, we have still an issue due to the wrong upgrade order (moveit-experimental being upgraded last):
https://hub.docker.com/r/moveit/moveit/builds/bxgqrlckusgbj9ta4a4ktr9

I prepared a new release, defining a (temporary) dependency from moveit-core to moveit-experimental:
https://github.com/ros/rosdistro/pull/19677

rhaschke, 2018-12-11:
With this latest release, both [DockerHub](https://hub.docker.com/r/moveit/moveit/builds/besz5vpufumiwyyrxihthht/) and [Travis](https://travis-ci.org/ros-planning/moveit/branches) are fine again. Release finished.

davetcoleman, 2018-12-11:
Thanks @rhaschke !

rhaschke, 2018-12-13:
Triggered a new release, https://github.com/ros/rosdistro/pull/19714, to fix regression #1259 / #1261.

rhaschke, 2018-12-17:
@clalancette Could you please hold back the upcoming ROS Melodic sync a few more days?
The recent MoveIt release introduced some regressions: https://github.com/ros-planning/moveit/pull/1271, https://github.com/ros-planning/moveit/issues/1255, which I would like to get merged / fixed before.

clalancette, 2018-12-17:
@rhaschke That's fine, I'm still waiting on the pilz_industrial_motion release before I go into freeze anyway.  If that gets done this week, but you still aren't ready, I'll start the freeze with an explicit exception for MoveIt.

rhaschke, 2018-12-24:
Triggered a new release 0.10.8: https://github.com/ros/rosdistro/pull/19825

rhaschke, 2018-12-29:
0.10.8 released and synced.

closed 2018-12-29 by rhaschke