moveit/moveit #1229: [[deprecated]]

v4hn, 2018-11-26:
Mentioned in https://github.com/ros-planning/moveit/pull/1146#issuecomment-441727608

Horray, we got C++14. Do we want to use [`[[deprecated]]`](https://en.wikipedia.org/wiki/C++14#The_attribute_[[deprecated]]) instead of [our own macros](https://github.com/ros-planning/moveit/blob/melodic-devel/moveit_core/macros/include/moveit/macros/deprecation.h)?

We never allowed for comments in our macros, so we would actually gain something from the change.

davetcoleman, 2018-11-26:
+1 to switching to the standard deprecated approach and removing custom macros!

rhaschke, 2018-11-28:
+1 from me too.

v4hn, 2018-11-29:
Although I don't believe this is a big problem, I want to point out that this will explicitly change our *header*-based standard requirement (in contrast to our own package-build dependency).
All user-based packages that include the headers will have to compile with c++14 support enabled.

v4hn, 2020-09-03:
Fixed by @NFiedler last year.