moveit/moveit #1230: How does the sequential BenchmarkExecutor.cpp program stop instances planning computations if they exceed the countdown?

moucrob, 2018-11-26:
### Description

Hello, 

in here https://github.com/ros-planning/moveit/tree/kinetic-devel/moveit_ros/benchmarks/src, 

one should call RunBenchmark.cpp, 

which itself calls BenchmarkExecutor.cpp (especially the method runBenchmark**s**(), and then runBenchmark() ).

My worry is that eventhough RunBenchmark.cpp inputs the moveit_ros_benchmarks::BenchmarkOptions to BenchmarkExecutor.cpp when calling it (including the timeout over which a plan computation should stop), **I do not see any spot where the computations are stopped**, except maybe within the ->solve() method. And I cannot find its source code...

So far, here are some URLs attempting to reach its source code : 
- http://docs.ros.org/jade/api/moveit_core/html/classplanning__interface_1_1PlanningContext.html#af3d95dd741609c58847bd312b8b033e0
- http://docs.ros.org/kinetic/api/moveit_core/html/structplanning__interface_1_1MotionPlanDetailedResponse.html
- http://docs.ros.org/kinetic/api/moveit_core/html/classplanning__interface_1_1PlanningContext.html

### Environment
* Kinetic

By any chance,

@rhaschke @beatrizleon @ryanluna @azompas could one of you enlighten me please?

By thanking you in advance

zkingston, 2018-11-26:
Within `RunBenchmark.cpp`, a `BenchmarkOptions` class is created which reads the desired timeout from "benchmark_config/parameters/timeout" (see [this line](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/benchmarks/src/BenchmarkOptions.cpp#L187)). Then, inside `BenchmarkExecutor`, these benchmarking options are distilled into a `moveit_msgs::MotionPlanRequest` in `BenchmarkExecutor::runBenchmark()` (see [this line](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/benchmarks/src/BenchmarkExecutor.cpp#L755)).

The `moveit_msgs::MotionPlanRequest` is then sent to the `planning_interface::PlannerManager` associated with each planner being benchmarked to retrieve a `planning_interface::PlanningContext` (see [this line](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/benchmarks/src/BenchmarkExecutor.cpp#L783)). The `planning_interface::PlanningContext::solve()` method is then called (see [this line](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/benchmarks/src/BenchmarkExecutor.cpp#L794)).

Different planner pipelines have different planning contexts, and handle the timeout conditions different. In the case of OMPL based planners, the timeout is finally specified within `ompl_interface::ModelBasedPlanningContext` (see [this line](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L645), in particular the line when the timeout is passed into OMPL [here](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L658)).

Each OMPL planner internally manages its timeout with a `ompl::base::PlannerTerminationCondition`. For example, within RRTConnect's source code you can see it being used in [its core loop](https://bitbucket.org/ompl/ompl/src/91e2ebe9afffb5a50106f33f9f2a95d5ec47610a/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp#lines-185), specifically [this line](https://bitbucket.org/ompl/ompl/src/91e2ebe9afffb5a50106f33f9f2a95d5ec47610a/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp#lines-232).

TL;DR: The timeout conditions are generally handled internally by each planner, and not by MoveIt! itself.

BryceStevenWilley, 2019-01-21:
@moucrob You can look at the [solve function](https://github.com/ros-planning/moveit/blob/32731d78be86b82e4445cd7222081b37b793c365/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L608), specifically at what is being added to the `description_` attribute. In this case, it's the planned trajectory, the simplified trajectory, and the finely interpolated trajectory.  Usually, the finely interpolated trajectory (the last one) will be the one executed by the robot.

moucrob, 2019-02-14:
@BryceStevenWilley Thank you !

I finally had a look in that `solve(`) function and have now some philosophical troubles with it then. I am not sure if this should be posted in this issue though.

It would be **about the "realtime-ness" of the ompl planners**.
Assume one calls a given one of your planners by that [`bool ompl_interface::ModelBasedPlanningContext::solve(planning_interface::MotionPlanDetailedResponse& res)`](https://github.com/ros-planning/moveit/blob/32731d78be86b82e4445cd7222081b37b793c365/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L599) function you gave me, in order to get a trajectory of robot states as answer, and few other things. Then in order to process further, one has to wait until a boolean is returned depending on the success of the computation.
So if one indeed had a (real-time, precise) processing time constraint, he would time from right before to right after the line calling `solve`.
In my case, I first thought the whole `solve` process wasn't violating the `request_.allowed_planning_time`, that I believe is through what most of the users input their needs. Which is wrong now that I dove in its code. 

- I can see that the `interpolateSolution()` internal post-process is imposed whereas it doesn't have an authoritarian switch/countdown. 

- Furthermore, in case `simplify_solutions_` is set to true, the timeout inputed in `simplifySolution()` doesn't take in account the few `push_back`, `resize`, `reset` preceding its call.

Should I definitively deal with using as recorded time the `res.processing_time_`, or do you guys plan to modify those parts of the code sooner or later?

closed 2019-02-15 by moucrob