moveit/moveit #1235: IK without "Allow Approximate IK Solutions" checked is too strict?

AIWintermuteAI, 2018-11-27:
### Your environment
* ROS Distro: |Kinetic]
* OS Version:  Ubuntu 16.04
* Binary build
* If binary, which release version? moveit_core 0.9.15

### Steps to reproduce
I have tried this both on my robotic arm for which I wrote urdf myself(https://github.com/AIWintermuteAI/ros-moveit-arm) and another robotic arm(https://github.com/jesseweisberg/moveo_ros). Without "Allow Approximate IK Solutions" checked in RVIZ I can barely move the arm's interactive markers. Rviz often becomes unresponsive for a couple of seconds. I have tried KDL and TRAC-IK  IK solvers, both exhibit same behavior. Using Python API to send the arm to a pose that theoretically should be reachable results in IK solver not being able to find a solution after 5 tries.
With "Allow Approximate IK Solutions" in RVIZ arm exhibits erratic behavior, but can be controlled with interactive markers. From my understanding  "Allow Approximate IK Solutions" cannot be enabled for Python API, so I cannot check if arm can go to the specified pose using Python API.

### Expected behaviour
The arm would go to a specified pose following the code in Python
```

    planned_grasp_data = plan_grasp(inpainted_color_image.rosmsg, inpainted_depth_image.rosmsg, camera_intrinsics, boundingBox)
    print(planned_grasp_data)
    grasp = planned_grasp_data.grasp
    # Specify a pose to place the target after being picked up
    target_pose = PoseStamped()
    target_pose.header.frame_id = REFERENCE_FRAME
    target_pose.pose.position.x = grasp.pose.position.x
    target_pose.pose.position.y = grasp.pose.position.y
    target_pose.pose.position.z = grasp.pose.position.z
    target_pose.pose.orientation.x = grasp.pose.orientation.x
    target_pose.pose.orientation.y = grasp.pose.orientation.y
    target_pose.pose.orientation.z = grasp.pose.orientation.z
    target_pose.pose.orientation.w = grasp.pose.orientation.w
    # Initialize the grasp pose to the target pose
    grasp_pose = target_pose
    print("going to target pose")
    right_arm.set_pose_target(target_pose)
    right_arm.go()
```
Here's the pose message
    position: 
      x: -0.0844764665968
      y: -0.0465208937066
      z: 0.576769530715
    orientation: 
      x: -0.237152548947
      y: -0.666152136173
      z: -0.237152548947
      w: 0.666152136173
  grasp_success_prob: 0.0483454577625

### Actual behaviour
[ERROR] [1543335424.629995768]: RRTConnect: Unable to sample any valid states for goal tree
[ INFO] [1543335424.630223583]: RRTConnect: Created 1 states (1 start + 0 goal)
[ INFO] [1543335424.630296220]: No solution found after 10.000725 seconds
[ INFO] [1543335426.800609033]: Unable to solve the planning problem
[image]

If necessary I can take a video of arm behavior in RVIZ with and without "Allow Approximate IK Solutions" checked.
Any input would be very much welcome. I'm trying to understand if it's a problem with my set-up or a limitation of MoveIt.

rhaschke, 2018-11-27:
Usually that's not an issue with MoveIt, but with your specific arm kinematics.
Looks like you start in a singular configuration, which might pose an issue. Do things change if you start from another configuration?
How many DoFs has your arm? I observed the described behaviour in the past for arms with only 5 DoF.

AIWintermuteAI, 2018-11-28:
The arm has 5 DOF. If that's an issue, are there any known workarounds for that? 
Can you elaborate on singular configuration issue? I am not sure if I understand what do you mean by that.

v4hn, 2018-12-04:
5 DOF arms cannot reach *almost all* 6D-Pose targets.
As a result, moving the 6D interactive marker around fails usually.

To solve this, use a dedicated IK solver (e.g. an IKFast plugin) that explicitly ignores the DOF it cannot reach, or use a generic optimizer (e.g. [bio_ik](https://github.com/TAMS-Group/bio_ik)) and rely on "good enough" solutions with Approximate IK.

Either way, *pose* requests for *arbitrary* poses will usually fail to fulfill the industry-required precision of 1e-5 (relative weighting position and orientation). So the user has to make sure they only request IK for targets that are actually reachable by the 5D arm.

Closing because this is commonly not really considered to be a problem of MoveIt, but of the arm kinematics.

closed 2018-12-04 by v4hn