moveit/moveit #1242: setup_assistant panel for joint_limits.yaml

v4hn, 2018-12-04:
The `joint_limits.yaml` file is generated by the setup assistant, but there is no way to edit the file from the assistant.

It would simplify setting limits if we could have sliders for optional position limits that move around the robot visualization too (just as for adding named states).
Joint velocity / acceleration limits could be simple edit-fields with checkboxes to enable/disable them.

To make this useful in practice, it should also be able to load the current `joint_limits.yaml` if there is one already.

MohmadAyman, 2018-12-09:
I could do that, after we get  #1243  merged.

SammyRamone, 2023-04-21:
This feature is still missing. Additionally, the setup assistant will ignore any current content of this file and overwrite it with default values (at least for the acceleration limits) when you change an existing moveit config.