moveit/moveit #1255: Bad IK results/issues via setFromIK AND RViz Motion Planning interactive marker (failures and/or joint flips)

BrettHemes, 2018-12-10:
### Description
Since migrating to Melodic existing code base has started failing with bad trajectories returned from custom interpolators leveraging RobotState::setFromIK calls.  See RosAnswers question [here](https://answers.ros.org/question/310062/incorrectunexpected-behavior-from-trac-ik-with-calls-to-moveitcorerobotstatesetfromik/) when I thought it was a Trac-IK issue.  After playing with multiple solvers ,the issue seems to be deeper within MoveIt.  Running the exact same trajectory results in random failures sometimes succeeding and sometimes failing.  The exact failure mode depends on the solver used and the corresponding parameters (e.g., timeout and number of attempts).

Trac-IK is in general more robust than KDL (making debugging a bit harder) but has issues with both Speed and Distance solve types.  With Speed type, the solver often succeeds but sometimes returns a solution significantly far away from the seeded pose (i.e., joint flip).  The trajectory is pretty dense and thus the seeds are of very "good" quality (e.g., displacement on the order of 1e-2 radians from the true solution) so I don't know why this is happening.  Same problem with Distance solve_type unless the time is set to an overly large number (e.g., > 5ms).  Even then failures still happen...

KDL is not returning joint flips per se but will randomly fail to return a valid solution.  If called multiple times in a row on the **exact same seed** it will succeed, so it seems very much random

**While investigating I found the issue is (I believe) reproducable by just dragging the interactive tool planning marker around in RViz (see Steps to reproduce below).**

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04
* Binary (newest as of today)

### Steps to reproduce
Calls to RobotState::setFromIK and/or dragging around interactive marker in RViz for any 6-dof industrial arms (e.g., kuka_experimental, fanuc, etc.).

Create a simple MoveIt Config via the setup assistant with the KR120 URDF from kuka_experimental.  Set chain solver from base_link to tool0 with either KDL or Trac-IK as the solver.  Run demo.launch and drag the interactive marker around.

### Expected behaviour
Continuous and good/close IK solutions.

### Actual behaviour
**With Trac-IK** small motions in the workspace of the interactive marker result in numerous and frequent joint flips (as was seen in API calls to RobotState::setFromIK).

**With KDL** the robot routinely gets "stuck" and will not respond to inputs.  Repeated grabs/releases of the marker will eventually result in a good IK solution.  **When stuck, any end effector geometries turn pinkish (vs normal green color).**  It seems a small rotation of the tool0 pose is more likely to get the model unstuck than a pure translational motion.

In case it offers any leads, **for both solvers, enabling "Allow approximate IK solutions" alleviates the problem**

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

BrettHemes, 2018-12-10:
[image]

Here is the KDL response of the marker just flying away from the flange while stuck...

BrettHemes, 2018-12-10:
Trac-IK example when translating the marker a small distance through a highly dexterous portion of the workspace showing a flip.
[image]
[image]

rhaschke, 2018-12-10:
Thanks for reporting this issue, which I can confirm. I also noticed recently that the tutorials' Panda arm is not smoothly following the rviz marker. I didn't have time yet to look into this and I'm not aware of any changes to the solvers. Did we changed any accuracy thresholds since Kinetic?

gavanderhoorn, 2018-12-12:
> I'm afraid that the transition from KDL 1.3 to KDL 1.4 causes this regression.

Do you suspect it, or did you actually figure out that the migration is the cause?

rhaschke, 2018-12-12:
I only suspect this. Tomorrow, I will try to test Melodic under Kinetic.

v4hn, 2018-12-13:
@BrettHemes Could you possibly provide feedback on whether moveit's `melodic-devel` branch still works for you on ROS kinetic (Ubuntu 16.04) too? (You will have to compile trac-ik locally too to test it)

Debugging this is painful, so the more information we have from the outside, the better.

v4hn, 2018-12-13:
Joint flips in Cartesian interpolation is a general problem of the method.

In theory the IK plugin should be informed about the path-planning application to avoid random-restarts methods in solving that create these flips.

I believe it is entirely possible that scheduling/timing in the individual components changed due to minor changes in the OS and it became (slightly) more likely that TracIK finds a solution independent from the seed first.

Long story short: I am not convinced this is a "regression" that can be fixed without functional changes in the trajectory generation.

rhaschke, 2018-12-13:
This issue is not only related to TracIK, but also present in the KDL plugin.
I agree with @v4hn that the Cartesian interpolation is not nicely designed in general, but it worked somehow in Kinetic. Now, it's more or less completely broken in Melodic.

To narrow it further down, I plan to build in a melodic-environement, but with KDL downgraded to 1.3.
However, I will only have time for this in the evening.

rhaschke, 2018-12-13:
I built in a Melodic environment with KDL 1.3. The good news is: It's not related to KDL, because the problem persists with KDL 1.3. Hence, it's some other 3rd-party lib in Bionic/Melodic causing the trouble.

BrettHemes, 2018-12-13:
> @BrettHemes Could you possibly provide feedback on whether moveit's melodic-devel branch still works for you on ROS kinetic (Ubuntu 16.04) too? (You will have to compile trac-ik locally too to test it)

@v4hn I tried a bit this morning getting a Kinetic Docker container going with nvidia support to no avail... got to where it "should" work but end up with segfaults from RViz...  My native installs are all now Melodic so I will have to get a VM going or find another machine with Kinetic on it.

rhaschke, 2018-12-13:
@BrettHemes Do you know about the [gui-docker script in the MoveIt source tree](https://github.com/ros-planning/moveit/blob/melodic-devel/.docker/gui-docker)? I used this to run rviz from docker. However, I used an Intel GPU, because I don't have [nvidia-docker](https://github.com/NVIDIA/nvidia-docker) installed.

rhaschke, 2018-12-13:
@BrettHemes I think, it's not necessary that you validate again that the melodic-branch works well on ROS Kinetic. I can reproduce the error and we just need to figure out which 3rd-party library is to blame.

BrettHemes, 2018-12-13:
> @BrettHemes I think, it's not necessary that you validate again that the melodic-branch works well on ROS Kinetic. I can reproduce the error and we just need to figure out which 3rd-party library is to blame.

Sounds good;  I appreciate the effort in looking into this quickly!

> @BrettHemes Do you know about the gui-docker script in the MoveIt source tree? I used this to run rviz from docker. However, I used an Intel GPU, because I don't have nvidia-docker installed.

I was not aware but used a similar approach based off stuff [here](https://github.com/osrf/docker_templates/issues/33) and other places.  I got everything working for Melodic but Kinetic needs extra steps to get libglvnd which I think is the culprit of my failure...

rhaschke, 2018-12-13:
I removed the image comments as they distracted me from finding relevant comments.
Let's summarize my rviz tests, i.e. loading `moveit_resources/fanuc_moveit_config/launch/demo.launch`, 
disabling collision-aware IK, and moving the arm around

| MoveIt             | on Kinetic     | on Melodic  |
| ------------------- |:----------------:| ---------------- |
| kinetic-devel      | OK             | smooth on translation, but stuck on rotation / after reaching singularity |
| melodic-devel    | OK             | no left-right translation, no rotation,  up-down OK |

rhaschke, 2018-12-14:
Working on unit tests now...

BrettHemes, 2018-12-14:
Also, in case it helps I have noticed that if an end effector is defined in the move group it turns "pinkish" when the chain gets stuck (versus its normal green color).  I looked through the visual tools code and only found constants for RED and GREEN.  Do you know what this pinkish color for the end effector designates or where it is getting set from?

simonschmeisser, 2019-01-03:
@rhaschke i did not manage to have a look at this before holidays, could you give a short update if this is fixed by one of your many PRs? Thanks a lot for your effort already!

rhaschke, 2019-01-03:
Unfortunately, I couldn't yet track down the issue. I have some improvements to the KDL solver in my pipeline though, but they are not yet filed as a PR. On some unit tests I could *not* find differences between Kinetic and Melodic (#1272).

rhaschke, 2019-01-04:
You mean the yaml-specified seed for the unit tests? Yes, these are correctly passed.

simonschmeisser, 2019-01-04:
I added a print statement in our ikfast plugin and it showed a seed I did not expect but there was too much code and too little time to really verify that

rhaschke, 2019-01-10:
I pushed several improvements to the KDL solver [to my fork](https://github.com/ubi-agni/moveit/tree/kdl-improvements). Although, I didn't find one single culprit resolving this regression, for me the suggested set of changes provides a really smooth and fast IK experience with KDL.

Before filing a PR for this, I'm awaiting for feedback on #1272, #1288, and #1294. But everybody affected by this regression is invited to test this fix and provide feedback.

rhaschke, 2019-01-10:
> Do you know what this pinkish color for the end effector designates or where it is getting set from?

@BrettHemes I also observed the pinkish color during my trials and I think it indicates joint-limit violations of the associate joint. Shouldn't actually happen with KDL solver.

BrettHemes, 2019-01-11:
I can maybe get to trying out the KDL improvements next week.  This still doesn't explain the associated behavior with TracIK and thus believe there is a different underlying issue.  I feel like an improperly set seed would explain very much the issues I have been observing (with both solvers).  It is probably worth investigating further @simonschmeisser's comment,

> I added a print statement in our ikfast plugin and it showed a seed I did not expect but there was too much code and too little time to really verify that

rhaschke, 2019-01-16:
I'm not using / testing ikfast, but the KDL solver uses (and used) the correct seed.
Maybe, you can elaborate a little bit more on your use case. Using the kinematics tests #1272, can you setup some unit tests, similar to the [Fanuc example](https://github.com/ros-planning/moveit/blob/afcf6b7b7d2463414d3d6e9b4318860df442f107/moveit_kinematics/test/fanuc.test)?

rhaschke, 2019-01-18:
The timeout defines how much time the IK solver might take to actually solve a request. If this value is too small, it cannot work in most situations. If your parameter was zero before, the solver gives up after the very first attempt.

emielke12, 2019-01-18:
Great! I think that the `kdl-improvements` branch provides the behavior we were expecting then.

simonschmeisser, 2019-01-22:
I might have mixed up a few things as during the transition from kinetic to melodic I also stumbled across  https://github.com/ros-planning/moveit/issues/1151 and added a enforcement of joint limits from `joint_limits.yaml` in our ik validity callback function. This prohibited some expected solutions. I did not spot obvious issues in the ikfast plugin code (besides dead code). As opw works just as good with way better maintainability I doubt I will spend much time on ikfast. So please skip my comments above.