moveit/moveit #1257: enhancement: support loading multiple planner plugins

gavanderhoorn, 2018-12-11:
### Description

It would be nice if MoveIt supported loading multiple planning plugins into a single instance of `move_group`.

This is currently not possible, requiring users to start multiple `move_group`s or use other tricks (like loading planner plugins directly into their own application).

rhaschke, 2018-12-11:
Which interface/API do you suggest to choose a specific plugin for planning?
Should we augment [MotionPlanRequest](http://docs.ros.org/melodic/api/moveit_msgs/html/msg/MotionPlanRequest.html) with a new field `planner_plugin` or should we hijack the existing `planner_id` and prepending the plugin name there, i.e. `<plugin name>/<planner id>`

gavanderhoorn, 2018-12-11:
I've done the last in the implementation that I linked.

I don't believe it's actually "hijacking": the semantics are exactly the same, as it specifies the specific planner to use. It just includes an additional "namespace".

davetcoleman, 2018-12-11:
I'm watching Gijs write this issue on his laptop from behind him, its amusing.

I like this proposal. In particular, I like the possibility that we could use a simple "local" planner first and if that fails, fall back to a "global" planner like RRT. This would require multiple planners be loaded at once. For example, I've been talking to the team behind [pilz_industrial_motion](https://github.com/PilzDE/pilz_industrial_motion) about integrating their Cartesian planner deeper into MoveIt!, and loading this and OMPL at once would be ideal. @martiniil @agutenkunst

gavanderhoorn, 2018-12-11:
Main issue with it right now -- part from the fact that it doesn't do any checking for anything -- is that it doesn't deal with the planning adapters.

Personally I'd like to have multiple sets of planning adapters loaded, one for each of the loaded planner pipelines.

Unfortunately that's a bit more involved than this simple meta planning plugin loader.

And most likely of course multi-planning plugin support should be implemented properly instead of with this hack.

gavanderhoorn, 2018-12-11:
I remember SwRI telling us that they've done something similar. Perhaps @Levi-Armstrong, @Jmeyer1292 and/or @jrgnicho can tell us something about that (in general, I imagine some things may be confidential)?

gavanderhoorn, 2019-11-15:
@henningkayser @davetcoleman: could you comment on what the status is of multi-planner loading in MoveIt?

I ask you, as #1530 could be seen as preperatory work and @henningkayser writes:

> @rhaschke wrote:
> > From the title, I was expecting a much larger contribution, namely supporting multiple planning pipelines (and planners) in the `move_group` node. To clarify the contribution of this PR, I changed the title.
> 
> I'm actually looking into this, but I think it's a first good step to be able to actually run different planning pipelines without using the MoveGroup node.

felixvd, 2020-03-10:
@henningkayser As not being able to mix "classic" PTP/LIN/CIRC motions (the `industrial_motion_planner` linked just above) and "free" motion planning (OMPL) would be hard to explain to users, what about adding this on the list of [potential code sprints](https://moveit.ros.org/documentation/contributing/future_projects/)? If you looked into it you might make a good mentor.