moveit/moveit #1263: controllers.yaml not loading properly

jav-solo, 2018-12-12:
### Description
When modifying the controllers.yaml I cannot get both controllers to load
Overview of your issue here.

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04

### Steps to reproduce
Use this controllers.yaml and load the controllers list. 

`controller_list:
 - name: ''
   action_ns: joint_trajectory_action
   type: FollowJointTrajectory
   default: true
   joints:
     - joint_1
     - joint_2
     - joint_3
     - joint_4
     - joint_5
     - joint_6
 - name: ''
   action_ns: gripper
   type: GripperCommand
   default: true
   parallel: true
   joints:
     - left_gripper_connection
     - right_gripper_connection
`

### Expected behaviour
It should load both controllers separately.

### Actual behaviour
It will only load the second one, whether it's joint trajectory action or gripper command. 

### Backtrace or Console output
`[ERROR] 1544654341.212747166: /move_group|ros.moveit_ros_planning.trajectory_execution_manager: Unable to identify any set of controllers that can actuate the specified joints: [ joint_1 joint_2 joint_3 joint_4 joint_5 joint_6 ]
[ERROR] 1544654341.212772597: /move_group|ros.moveit_ros_planning.trajectory_execution_manager: Known controllers and their joints:
controller '' controls joints:
  left_gripper_connection
  right_gripper_connection
`
Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

gavanderhoorn, 2018-12-13:
You appear to be loading two controllers with the same name -- which is also an empty name.

Try to give your controllers different names. That should make things work.

jav-solo, 2018-12-13:
So, for some reason it could not find the controllers if I simply added a different name to each. However what did work is swapping the name with the action_ns I was using for each as such:

`controller_list:
 - name: joint_trajectory_action
   action_ns: ''
   type: FollowJointTrajectory
   default: true
   joints:
     - joint_1
     - joint_2
     - joint_3
     - joint_4
     - joint_5
     - joint_6
 - name: gripper
   action_ns: ''
   type: GripperCommand
   default: true
   parallel: true
   joints:
     - left_gripper_connection
     - right_gripper_connection`

closed 2018-12-13 by jav-solo