moveit/moveit #1264: CHOMP can't plan path avoid obstacle if using mimic joints

Yosukee, 2018-12-13:
I tried CHOMP using mimic joints, but CHOMP can't plan path avoid obstacle.
To use 6dof robot as 4dof, I change 2 joints to mimic joint.
When planning CHOMP show these error messages.
If I use robot(6dof) used not mimic, CHOMP can plan avoid obstacle.

[image]

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04
* Robot: mh180-120 (http://wiki.ros.org/motoman_mh_support)

rhaschke, 2018-12-13:
A 4dof robot cannot solve the IK problem anymore (which is 6dof)! This is not chomp-related, but a design-problem on your side.

closed 2018-12-13 by rhaschke

Yosukee, 2018-12-13:
I think this is not the case.
OMPL (with LMAKinematicsPlugin kinematics solver) works perfectly with 2 mimic joints for my 6DOF robot. It can avoid obstacles, and no errors at all. Mimic joints are used in order to plan in 4DOF with OMPL, but in the end 6DOF robot is displayed in RViz. Maybe I can post a video later, too.
If the planner is changed to CHOMP, simple motion planning is successful, but unfortunately if obstacles are added to the planning scene, motion planning almost always fails. I think it is not a design problem, since OMPL works without any problem. 
What do you think? Thank you for your help in advance!

rhaschke, 2018-12-13:
OMPL usually uses joint-space sampling. Thus, because no IK is involved, it can handle underactuated situations - as long as the goal is feasible.
I'm not sure about the approach in CHOMP, but from your screenshot I infer that CHOMP is using IK.
Maybe @bmagyar can provide more insight.

bmagyar, 2018-12-18:
I agree with @rhaschke , I think there is a feasibility issue here. There may still be a solution to the problem even in the 4DoF case but probably the problem is very hard and thus CHOMP fails to optimize it's way out of it - after all, any optimization-based solution is prone to local minimas.