moveit/moveit #1268: Reconfiguring planner parameters online

adam-vonderviszt, 2018-12-15:
### Description

In short, I need to change planner parameters like longest_valid_segment_fraction online. Using "set_planner_params" service I can change it, and with "get_planner_params" service I can also verify the change, but after sending a planning request the following message appears:

`[ INFO] [1544851647.956426953]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1544851647.958932473]: Planner configuration 'manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.`

Is there a way to change planner parameters or just reconstruct the planner online?

### Environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Binary

adam-vonderviszt, 2018-12-16:
Yes, the parameters are not applied at all.
If the longest_valid_segment_fraction parameter is changed to a very low value in the config file before launching Moveit, motion planning takes a lot of time as it should, but if it is changed only after launching Moveit using the set_planning_params service, the change is not applied, and the planning is fast as usual.

rhaschke, 2018-12-17:
I just verified the behaviour, both in Kinetic and Melodic, and it works like a charm: Changing the parameter `longest_valid_segment_fraction` from the rviz MotionPlanning panel, I can adjust the smoothness of the generated trajectory - as expected. To validate this on your side, please install package `ros-kinetic-moveit-resources` and run:
`roslaunch /opt/ros/kinetic/share/moveit_resources/fanuc_moveit_config/launch/demo.launch`

If this is working for you (which I am sure it is), your issue is a configuration problem.

adam-vonderviszt, 2018-12-17:
You are right, changing the parameter and planning in Rviz works for me, too.

The problem happens, if planning is executed in python code using Moveit Commander group.plan() function. Even if `longest_valid_segment_fraction` is setted in Rviz or `set_planner_params`, the change is not applied to python motion planning. Dont know what I am missing... Even though, the change can be verified by `get_planner_params` service in the same python code.

rhaschke, 2018-12-17:
Is it possible that the python wrapper reset the parameters? Could you check before and after planning with `get_planner_params`?

adam-vonderviszt, 2018-12-18:
The parameter before and after planning is the same, so python wrapper doesnt reset it, just somehow not applied.

adam-vonderviszt, 2018-12-18:
Ok, I managed to make it work!
The trick was I need to call group.set_planner_id() before planning to apply the changes made by Rviz or the python code.

adam-vonderviszt, 2018-12-18:
To avoid these kind of problems in the future, I think it would be reasonable to set the planner id to the default planner at initialization stage.