moveit/moveit #1269: Dramatically Reduce Memory Copy with OMPL

davetcoleman, 2018-12-15:
@v4hn requested I create an issue on a optimization I did in graduate school that has never made it back into the mainline MoveIt!. The reason is because it depends on the robot being used and whether the joint model group used (the planning group) has its joint values in a continuous array. And also I guess its not a fully complete idea, as I can't remember all the details at this point.

MoveIt! allows arbitrary planning groups to be created so, for example, you could have ``arm1`` defined as ``[joint1, joint3, joint4]``, which would not be continuous in the ``moveit::core::RobotState`` memory. 

*However*, if the memory is contiguous... which it normally is... you can change the ``ompl::State`` to be a template with ``N`` the number of DOF. Then you can store a large array of states in continuous memory as shown in [allocStates](https://github.com/PickNikRobotics/bolt/blob/kinetic-devel/bolt_moveit/include/bolt_moveit/model_size_state_space.h#L84).

With this optimization, if you need to quickly insert a lot of states into a graph you can make one allocation call instead of one of each generated state added to an array.

There are a lot of other possible optimizations also. For example, in MoveIt!'s current [OMPL State](https://github.com/ros-planning/moveit/blob/melodic-devel/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/parameterization/model_based_state_space.h#L165) none of the memory is necessary except the ``double values*`` array ([updated approach here](https://github.com/PickNikRobotics/bolt/blob/kinetic-devel/bolt_moveit/include/bolt_moveit/model_size_state_space.h#L58)).

gavanderhoorn, 2018-12-15:
> MoveIt! allows arbitrary planning groups to be created so, for example, you could have `arm1` defined as `[joint1, joint3, joint4]`, which would not be continuous in the `moveit::core::RobotState` [...]

Might be worth it to see whether we can create some sort of "view" structure / intermediate mapping that allows one to *always* have contiguous allocation of structures but then whenever needed allows for arbitrary access / mapping of accessors to this underlying structure.

That would -- conceptually at least -- seem to always support this sort of optimisation instead of only in particular cases.

rhaschke, 2018-12-16:
@davetcoleman Unfortunately, the links you provided are broken.

davetcoleman, 2018-12-31:
I've fixed the deadlinks @rhaschke