moveit/moveit #1273: Potential out-of-bounds access at moveit setup assistant

sharronliu, 2018-12-17:
### Description
moveit/moveit_setup_assistant/src/widgets/robot_poses_widget.cpp:RobotPosesWidget::loadJointSlider()

"&init_value" is passed to "getVariableDefaultPositions()" which uses it as an array, potential out-of-bounds access.

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04

### Steps to reproduce
code review

davetcoleman, 2018-12-21:
Good catch! Do you have a suggested solution?

rhaschke, 2019-01-02:
> `RobotPosesWidget::loadJointSlider()` passes `&init_value` to `getVariableDefaultPositions()`

It's [checked in advance, that the joint uses only a single variable](https://github.com/ros-planning/moveit/blob/4d25c3a8e9de614afa28b234dcc6c83648bc47f4/moveit_setup_assistant/src/widgets/robot_poses_widget.cpp#L519), i.e. a one-element array. 

> `RobotPosesWidget::showDefaultPose()` passes `&init_value` to `getVariableDefaultPositions()`

[Same here](https://github.com/ros-planning/moveit/blob/4d25c3a8e9de614afa28b234dcc6c83648bc47f4/moveit_setup_assistant/src/widgets/robot_poses_widget.cpp#L367).

closed 2019-01-02 by rhaschke