moveit/moveit #1278: MoveGroupInterface::pick() always failed at IKConstraintSampler::callIK() due to random seed is used

sharronliu, 2018-12-17:
### Description
Use MoveGroupInterface::pick() and the motion planning always failed at IKConstraintSampler::callIK(), due to "FALSE" is always passed via the last param "use_as_seed".

As comparison, when the same target pose passed to MoveGroupInterface::setPoseTarget(), the motion planning always success.

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04

### Steps to reproduce
// code section to use MoveGroupInterface::pick(), motion planning always failed at IKConstraintSampler::callIK() due to random seed is used
void pick(moveit::planning_interface::MoveGroupInterface& move_group)
{
  std::vector<moveit_msgs::Grasp> grasps;
  grasps.resize(1);
  grasps[0].grasp_pose.header.frame_id = "**base**";
  **grasps[0].grasp_pose.pose.orientation.y = -1;
  grasps[0].grasp_pose.pose.position.x = 0;
  grasps[0].grasp_pose.pose.position.y = -0.5;
  grasps[0].grasp_pose.pose.position.z = 0.1;**

  grasps[0].pre_grasp_approach.direction.header.frame_id = "base";
  grasps[0].pre_grasp_approach.direction.vector.z = -1.0;
  grasps[0].pre_grasp_approach.min_distance = 0.08;
  grasps[0].pre_grasp_approach.desired_distance = 0.1;

  grasps[0].post_grasp_retreat.direction.header.frame_id = "base";
  grasps[0].post_grasp_retreat.direction.vector.z = 1.0;
  grasps[0].post_grasp_retreat.min_distance = 0.08;
  grasps[0].post_grasp_retreat.desired_distance = 0.1;

  move_group.**setStartStateToCurrentState**();
  move_group.**pick**("object", grasps);
}

// code section to use MoveGroupInterface::setPoseTarget(), motion planning success when same target pose passed.
void move(moveit::planning_interface::MoveGroupInterface& move_group)
{
  move_group.**setStartStateToCurrentState**();
  move_group.setPoseReferenceFrame("**base**");
  geometry_msgs::Pose target_pose1;
  **target_pose1.orientation.y = -1.0;
  target_pose1.position.x = 0;
  target_pose1.position.y = -0.5;
  target_pose1.position.z = 0.1;**
  move_group.**setPoseTarget**(target_pose1, move_group.getEndEffectorLink());
  moveit::planning_interface::MoveGroupInterface::Plan my_plan;
  bool success = (move_group.plan(my_plan) == moveit::planning_interface::MoveItErrorCode::SUCCESS);
  ROS_INFO("plan pose goal %s", success ? "SUCCESS" : "FAILED");
} // ---> motion planning returns success for this pose.

### Expected behaviour
IKConstraintSampler::sampleHelper() should return success, since MoveGroupInterface::setPoseTarget() return success when the same target pose passed.

### Actual behaviour
IKConstraintSampler::callIK() always return failed, due to "FALSE" is always passed via the last param "use_as_seed".

### Backtrace or Console output

sharronliu, 2018-12-17:
Additional inputs, if "use_as_seed", the last param of IKConstraintSampler::callIK(), was set as "TRUE", the motion planning become success.

It deserved to investigate why "FALSE" is always passed when doing a "MoveGroupInterface::pick()".

felixvd, 2018-12-17:
Could you link the part in the code where the parameter `use_as_seed` passes the wrong value, as you say?

sharronliu, 2018-12-17:
> 
> 
> Could you link the part in the code where the parameter `use_as_seed` passes the wrong value, as you say?

https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/constraint_samplers/src/default_constraint_samplers.cpp#L587
if (callIK(ik_query, adapted_ik_validity_callback, ik_timeout_, state, **project && a == 0**))

felixvd, 2018-12-17:
Could you also link where `project` is passed through, for `pick()` and `setPoseTarget()` respectively? I haven't dealt with this part of the program a lot. Thanks.

sharronliu, 2018-12-17:
"project" is not passed from user app, instead it was internally handled by the moveit core. See here https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/constraint_samplers/src/default_constraint_samplers.cpp#L526

Changing the "project" (which looks like specific for "sample" purpose) may not be a safe fixing. A possible workaround that verified OK on my side is to have "FALSE" passed when "a=0". Any idea if this's a good fixing?
if (callIK(ik_query, adapted_ik_validity_callback, ik_timeout_, state, project **||** a == 0))

felixvd, 2018-12-17:
I'm not sure that I understand where the difference is right now, so that would be great. Are you sure that there is no other reason for `pick()` failing except for this parameter? Different planning times, for example?

sharronliu, 2018-12-17:
No other reason. In my investigation, the only change that may have "pick()" success is to pass "TRUE" to "use_as_seed"

felixvd, 2018-12-17:
How did you test that precisely? And is the solution that is found by `pick()` correct and feasible when you set `use_as_seed` to `True`?

sharronliu, 2018-12-17:
I once change the following params, which do not help:
- increase the time out
- increase the attempt counts

The motion planning can be executed successfully when setting "use_as_seed" to true from this way:
if (callIK(ik_query, adapted_ik_validity_callback, ik_timeout_, state, project **||** a == 0))

The "||" makes "use_as_seed" true when "a==0", which is the initial attempt. That implies, use the current robot state as seed at the initial attempt.

felixvd, 2018-12-17:
Could you explain how that happens and where? `sampleHelper` is internal to the `IKConstraintSampler` class. How do `pick()` and `setPoseTarget()` interact with the constraint sampler differently?

Note that `pick()` does not fail systematically, it only seems to have a low success rate. The [tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pick_place/pick_place_tutorial.html) code and some other instances failed often for me, but when I filled the vector with 20 entries of the same Grasp, some of the attempts succeeded.