moveit/moveit #1281: Using rosdep (rdmanifests) to install a "devel" build of moveit

gavanderhoorn, 2018-12-19:
This came up in yesterday's maintainer meeting and I thought I'd posts some info here.

I remember I asked a question on ROS Answers about the status of `rdmanifests` (the 'official' name for what @jonbinney referred to): [#q255238](https://answers.ros.org/question/255238).

See the question itself for more details, but @tfoote answered with this:

> They are a tool that works for developers who are installing from source on more exotic platforms. But it's an important tool for users not on our main supported platforms.
> 
> They are not recommended for significant distribution and sharing. And cannot be used in a released package.
> 
> We don't expect to put a lot of energy into adding features. But because there are use cases where it is valuable and there's no viable alternative, we have not chosen to deprecate it.
> 
> My recommendation is that rdmanifests should only be used as a last resort if there are no other options.

jonbinney, 2018-12-19:
We've been using an rdmanifest rule to install the Basler pylon camera library on our dev computers and robots, and it has worked pretty well. It actually just downloads and installs a deb, but used to download and unpack a targall. The rule is here https://github.com/magazino/pylon_camera/blob/indigo-devel/rosdep/pylon_sdk.rdmanifest

gavanderhoorn, 2018-12-19:
> We've been using an rdmanifest rule to install the Basler pylon camera library on our dev computers and robots, [..]

yes, I guessed you ran into them with the Pylon package.

That is also the package I used as an example in the ROS Answers question I linked to in the OP.

> and it has worked pretty well.

sure.

> It actually just downloads and installs a deb, but used to download and unpack a targall.

yes. And all as root iirc. Not sure that is something we'd want to use for distribution of moveit.

jonbinney, 2018-12-19:
Agree that it's a bit sketch to run random stuff as root. A deb we build isn't strictly better though, is it? Can't debs have arbitrary post install scripts that get run as root by dpkg?

gavanderhoorn, 2018-12-20:
> A deb we build isn't strictly better though, is it? Can't debs have arbitrary post install scripts that get run as root by dpkg?

true. Technically that could do the same things, although I believe the assumption is that debs go through something of a review process -- but that could of course also be the case for `rdmanifest`s.

tfoote, 2018-12-21:
The trust is approximately the same for adding the rdmanifest into your rosdep sources as adding a new apt repository. They both can have a requirement to trust that maintainer. We do at least force the md5sum check for the rdmanifest but that's definitely not foolproof these days. And with extra network resources there's other possible compromises. Though yes, some debs do things like pull external resources too. The biggest advantage of debs is that they ususally uninstall better and are less likely to have errors in detecting that they're already installed etc. 

If you're going to go to this level of effort I'd suggest actually bulding a deb. They're not that hard to build, and there's a lot of tools that can do a lot of the heavy lifting.