moveit/moveit #1284: Migrate moveit_tutorials into main moveit repo

davetcoleman, 2018-12-21:
I think our documentation would be much easier to maintain if feature changes could be combined with documentation changes in the same PR. For example, we often see comments like [this](https://github.com/ros-planning/moveit/pull/1253):

> merged I'll try to get around to add the tutorial soon.

This is a perfectly natural comment for anyone to make who is focused on delivering good code and features for their project. However, this is not ideal for the documentation which quickly becomes out of date or has bad coverage. 

I'm proposing we create a mechanism to combat human nature's tendency to put off the 'boring' task of documentation by requiring it co-exist in every pull request before having it merged in. Many PRs won't need to change documentation because they won't change behavior, which is fine. But others will.

I did some analysis on what this would take just now:
- [ ] Setup [gitlfs](https://git-lfs.github.com/) for moveit so that large asset files (images) don't clog the git repo
- [ ] Document how to use gitlfs for new users
- [ ] Import the moveit_tutorials git history into the moveit repo
- [ ] Combine the ``.travis.yml`` file from moveit_tutorials somehow into moveit_ci. The current moveit_tutorials Travis file is pretty slick in that it checks for proper sphynx format, checks for dead hyperlinks, and it auto-deploys the updated documentation to Github pages instantly. 

Thoughts?

gavanderhoorn, 2018-12-22:
I'm not entirely sure I understand how merging the documentation into the main repository makes it easier to keep things in-sync: if someone submits a PR without updating documentation, it doesn't really seem to matter whether the PR to `ros-planning/moveit` is missing the doc updates, or there isn't one against `ros-planning/moveit_tutorials`.

And similar to what we discussed in the last maintainer meeting: requiring doc updates before merging PRs would seem to lead to more iterations before a PR gets merged. In my experience that does not streamline the process, but actually makes it harder (you're creating more hurdles for contributors to deal with).

Merging the docs in will lead to an even larger repository, with more packages and potentially longer build times (I'm aware of the fact that we could make building docs optional). Not sure that would be a positive thing.

As an alternative: we could instate a procedure where a PR against the latter must exist before a PR against the former will be accepted.

v4hn, 2018-12-25:
This proposal needs to be discussed at a maintainer meeting.

Also, thank you for quoting me @davetcoleman :stuck_out_tongue: 

I mostly agree with @gavanderhoorn. I don't think it is an acceptable hurdle to require tutorial patches when people don't have the enthusiasm or time available to do them.

Also, I don't like the idea of using gitlfs for our main repo, although I quite like the tool for lab-internal projects.

rhaschke, 2019-01-02:
I agree to @gavanderhoorn: Merging the tutorials into the main repo doesn't facilitate the need to prepare corresponding documentation changes for a code PR. Maybe, it's a little bit easier to keep track of these coordinated PRs. However, this doesn't justify all the cons you already listed.
I think having a policy, as suggested by Gijs, to have a documentation PR before a code PR can be accepted is fully sufficient.

rhaschke, 2019-01-16:
I don't like the idea to further proceed along the route towards a monolithic MoveIt repository. We already merged most source code repos into a central one and it definitively facilitates coordinated source PRs. However, it also slows down CI cycles dramatically, compared to many more-or-less independent repos.
For this reason I vote against merging. If you decide for merging anyway, I strongly discourage usage of gitlfs, because this will impede users to contribute nicely illustrated tutorial pages.

rhaschke, 2019-01-21:
Another argument against merging is that we often want to hold back tutorial changes until features are actually released. An example for this, @davetcoleman gives himself in https://github.com/ros-planning/moveit_tutorials/pull/272#issue-240585729:
> Should not be merged until ... is merged and the next release is made.

v4hn, 2019-01-21:
Looking at the comment history here, everyone who replied here except for Dave opposes the idea. (Sorry Dave)

I will close the issue, but feel free to bring the topic back up again.

mlautman, 2019-01-22:
I have been devoting a good amount of time to moveit_tutorials over the last year and based on that experience, strongly agree with @davetcoleman here. The key issue I see with the tutorials is that they are treated as an afterthought. Moving the tutorials into the moveit repo seems like an obvious way to address this.

- Adding the tutorials to CI for moveit will be much more effective at encouraging new contributors to update the tutorials than any new policy. Making PR's against multiple repos is a pain. By combining moveit_tutorials into moveit we will reduce the barrier to documenting new contributions in the tutorials significantly.
- In the last few months I have seen a number of issues where users failed to compile moveit_tutorials against their released version of MoveIt. (eg. https://github.com/ros-planning/moveit_tutorials/issues/262) Combining repos solves this by ensuring that the tutorials and moveit are released together and built together. 
- I agree that it is annoying that gitlfs doesn't work out of the box. That said, it is really easy to set up and adding instructions to the tutorials is trivial. Besides, we don't actually have large enough files in the tutorials to need gitlfs. Adding moveit_tutorials would add 25MiB to a 55MiB repo. Not that big of a deal. 
- As for compile times, the moveit_tutorials package takes 30sec on my machine. For comparison, the moveit_setup_assistant takes 45 seconds and moveit_ros_warehouse takes 1 min. If we really want to speed up average compile times, we should add documentation teaching new users to use `catkin config --blacklist`. Compile time is not an an argument for keeping the project spread across different repositories.

rhaschke, 2019-01-22:
> Adding the tutorials to the moveit repo will be much more effective at encouraging new contributors to 
update the tutorials than any new policy.

There were many arguments against this before and you don't provide a new pro argument.

> In the last few months I have seen a number of issues where users failed to compile moveit_tutorials against released MoveIt binaries.

Combining repos essentially will enforce people to build the whole MoveIt source, which takes more than 30min! That's not the first-time user experience you want to convey either. 
To avoid this issue (which I consider serious as well), we should keep separate repos, but collect feature-related changes to the tutorials on a separate branch. These changes should only be merged after a successful MoveIt release of those new features. This also requires a new Travis policy: The `melodic-release` branch should be build against released sources, while the `melodic-devel` branch should be build against latest MoveIt melodic-devel.

davetcoleman, 2019-01-22:
> We should add documentation teaching new users to use catkin config --blacklist

I [did that](https://github.com/ros-planning/moveit.ros.org/commit/004734738cfbe32b15036acac752d92302b11577#diff-4e11f0972a6c6443e6e0925a94e6433c) a while back ;-) 

> Combining repos essentially will enforce people to build the whole MoveIt source, which takes more than 30min! 

This is a decent point I hadn't thought about... how necessary is it for beginners to actually download the tutorials, vs just copy paste code into their own files? I know it does currently advocate this on the Getting Started page. Still, blacklisting could address this

> but collect feature-related changes to the tutorials on a separate branch

Assuming we don't merge these repos, I think we should have a ``master`` branch for the tutorials, just like we plan to do for the main repo. We in fact already somewhat handle this issue with the following note to users on the top of every tutorials page:

> You are on the latest ROS Melodic version of the tutorials, which is less stable. For beginners we recommmend the more stable ROS Kinetic tutorials (Requires Ubuntu 16.04). For older computers on Ubuntu 14.04 see ROS Indigo tutorials with the PR2.

> Adding the tutorials to the moveit repo will be much more effective at encouraging new contributors to
update the tutorials than any new policy.

I think this point from @mlautman should not be taken lightly. Mike and I have by far done the most work on the moveit_tutorials (70 and 59 commits, respectively) and I also put a lot of work into the old moveit wiki. Reducing the friction for documentation is a really important thing. Its why many projects just have a large README in the root of their repo. Wikis are also much easier than having to deal with juggling multiple pull requests. I think the thing we should most optimize for is MoveIt! developers being able to easily add docs to their new features as they make them.

rhaschke, 2019-01-22:
> blacklisting could address this

I don't see how blacklisting solves the issue raised by @mlautman: Building only the tutorial and blacklisting MoveIt will essentially again build against the binary release and thus cause errors.

Regarding your main argument for merging ("to facilitate creation of docs for new features"), I don't see any new argument. There is not more or less pressure on developers to document their stuff, when the tutorials are merged. It's our task as maintainers to ensure that PRs for new features are accompanied by corresponding documentation PRs. As I noted above, these typically should be slightly delayed, such that the tutorials build nicely against the released binary packages.