moveit/moveit #1297: Localization issues in OMPL

simonschmeisser, 2019-01-08:
### Description

I recently fixed how MoveIt! hands over float/double parameters to OMPL (ie using "." as a decimal separator independent of your system locale). Now it turns out that at about the same time @mamoll changed the code in OMPL to expect localized parameters (ie "longest_valid_segment_fraction: 0,2"). So if you get an error  
```
[ INFO] [1546970381.535683436]: Planner configuration 'manipulator[RRTConnectkConfigDefault]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ERROR] [1546970381.535761953]: OMPL encountered an error: The fraction of the extent must be larger than 0 and less than 1

```
this is the reason. (std::stof/stod will interpret "0.2" as "0" as it stops at the first unknown character)

I opened an issue at https://bitbucket.org/ompl/ompl/issues/496/localization-errors-or-the-tale-of-decimal

I will check tomorrow how to do pull requests on bitbucket unless @mamoll beats me to it

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04
* MoveIt!: source (current), OMPL: binary (1.4.1 2018-12-06)

simonschmeisser, 2019-01-09:
basic PR is at https://github.com/ompl/ompl/pull/112

simonschmeisser, 2019-01-09:
oh and if somebody else encounters this, you can workaround by just not specifying "longest_valid_segment_fraction" in config/ompl_planner.yaml but be warned that other real value parameters will likely be truncated as well

mamoll, 2019-01-09:
Fixed in OMPL's default branch in https://bitbucket.org/ompl/ompl/commits/25d8eb8e75a177e9a35a3af264b1224330d2fa29.

BryceStevenWilley, 2019-04-26:
@simonschmeisser Looks like the OMPL fix made it in the [Melodic sync](https://discourse.ros.org/t/new-packages-for-melodic-2019-02-11/7809) in February, can you confirm that this issue is fixed?

closed 2022-11-20 by simonschmeisser