moveit/moveit #1302: no motion plan found

C-monC, 2019-01-10:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Indigo|Jade|Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2019-01-10:
According to your console output, a solution is found and successfully executed using the fake controller:
```
[ INFO] [1547123876.198651041]: Solution found in 0.013047 seconds 
[ INFO] [1547123876.209667728]: SimpleSetup: Path simplification took 0.010899 seconds and changed from 3 to 2 states 
[ INFO] [1547123876.284230017]: Fake execution of trajectory 
[ INFO] [1547123878.683883080]: Completed trajectory execution with status SUCCEEDED
```
What is the actual problem?

C-monC, 2019-01-11:
I can plan useing set joint states but I can't execute the path in Rviz.
I can't plan using pose goal or execute using pose goal. `^[[ INFO] [1547187664.749495665]: ABORTED: No motion plan found. No execution attempted.
`
I used get current pose to get `position: 
  x: 0.102592509978
  y: 3.56171850966e-05
  z: 0.156007748445
orientation: 
  x: -0.603189755716
  y: -0.369006495064
  z: -0.368769278832
  w: 0.603328719845
`
but setting the pose using ` pose_goal = []
    pose_goal = [0.4, 3, 0.1]
    move_group.set_position_target(pose_goal)`

gives TypeError: 'Pose' object does not support indexing

C-monC, 2019-01-11:
If in the docs it says the function wants xyz is that a list?

rhaschke, 2019-01-11:
You should read the MoveIt tutorial: pose goals with Python are explained [here](https://ros-planning.github.io/moveit_tutorials/doc/move_group_python_interface/move_group_python_interface_tutorial.html#planning-to-a-pose-goal). Please raise such usage questions on https://answers.ros.org.
Closing this, as it's obviously not an issue, but a usage question.

closed 2019-01-11 by rhaschke