moveit/moveit #1303: DisplayTrajectory RViz Plugin Memory Leak

wxmerkt, 2019-01-10:
### Description
Displaying `DisplayTrajectory` with show trail takes 100% CPU and increases memory usage into the Gigabytes - even after hiding/disabling the plugin. Memory usage increases with number of sent messages.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary,  `0.10.8-0bionic.20181227.060642`

### Steps to reproduce
- Publish DisplayTrajectory message (e.g., with ~100 waypoints) and activate `Show Trail`
- Publish a few of those and notice that CPU will be at 100% and memory usage creep up into the GB.

rhaschke, 2019-01-10:
This is a duplicate of https://github.com/ros-visualization/rviz/issues/1222, which is resolved by https://github.com/ros-visualization/rviz/pull/1294 (but unfortunately not yet merged).

closed 2019-01-10 by rhaschke

wxmerkt, 2019-01-10:
Thank you for the quick response and pointer - I really hope that rviz fix is getting merged soon.