moveit/moveit #1304: Setup Assistant 2.0 URDF for Gazebo Not Working

superxingzheng, 2019-01-11:
### Description

I am trying to simulate the Franka Panda robot used in Moveit tutorials in Gazebo and move it with the example codes from the tutorials. I think this is going to be a missing part for most of beginners of Moveit. The Setup Assistant seems to be a promising tool based on [this](https://discourse.ros.org/t/google-summer-of-code-improvements-to-setup-assistant/6659) post.

But from what I see from trying it, the new tabs in the SA 2.0 are far away from being ready.
1. The <inertial> elements inserted into the URDF for Gazebo are wrong. They are inertia matrices of boxes. They should be cylinder inertia matrices. You can see this by turning on the view of inertia in Gazebo.
2. The fixed virtual joint for mounting the robot to the ground is not working. The robot fails to the ground if you unpause Gazebo.
3. The transmissions that are necessary for Gazebo to simulate hardware are missing. I am curious about how was [this](http://moveit.ros.org/assets/images/gsoc18/simulation.gif) made working without the transmissions?

### Your environment
* ROS Distro: [Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Binary

gavanderhoorn, 2019-01-11:
>     2\. The fixed virtual joint for mounting the robot to the ground is not working. The robot fails to the ground if you unpause Gazebo.

Just an obervation/comment: MoveIt's fixed joint is *not* a urdf/sdf `fixed` joint. It will not attach your robot to Gazebo's world frame fi.

So your robot falling has a different cause.

superxingzheng, 2019-01-11:
Step 3 of the [tutorial of the Setup Assistant](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html) is for adding this fixed virtual joint. 

I assume when the SA insert elements into the URDF for Gazebo, it should insert a fixed joint to mount the robot to the ground.

superxingzheng, 2019-01-11:
Just to leave one thought here. Rather than trying to automatically create a URDF for any industrial robot to be working in Gazebo, integrating with the [ROS-Industrial](https://github.com/ros-industrial/) projects seems to be more realistic. For example the [UR5](https://github.com/ros-industrial/universal_robot) seems to be already working. Annoyingly, the UR5 does not have a gripper...

gavanderhoorn, 2019-01-11:
> Step 3 of the [tutorial of the Setup Assistant](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html) is for adding this fixed virtual joint.

Yes, but that is a MoveIt feature. It does not influence what gets loaded by Gazebo (or any other consumer of the URDF/xacro, for that matter).

> I assume when the SA insert elements into the URDF for Gazebo, it should insert a fixed joint to mount the robot to the ground.

I don't believe that happens, but in any case, the "fixed" joint added in step 3 of the tutorial you link is not a `fixed` joint in the urdf.

superxingzheng, 2019-01-11:
I realized I have not deleted the URDF generated the by the SA. 

> I don't believe that happens, but in any case, the "fixed" joint added in step 3 of the tutorial you link is not a fixed joint in the urdf.

That does not happen. There is no fixed joint between the world and the base of the robot int the urdf.

MohmadAyman, 2019-01-26:
> The elements inserted into the URDF for Gazebo are wrong. They are inertia matrices of boxes. They should be cylinder inertia matrices. You can see this by turning on the view of inertia in Gazebo.

While working on this we decided to make simulating in gazebo easier, trying to detect the inertia of a shape from an `stl` file seemed a little over killing, so we followed the suggestion in the [OSRF tutorials to use a small inertia value.](http://wiki.ros.org/urdf/Tutorials/Adding%20Physical%20and%20Collision%20Properties%20to%20a%20URDF%20Model)

> The transmissions that are necessary for Gazebo to simulate hardware are missing. I am curious about how this was made working without the transmissions?

They are put at the bottom of the simulation screen after you generate the `urdf` before the closing of the `robot` tag.

superxingzheng, 2019-01-28:
@MohmadAyman 
Thanks! I see the transmissions when using the SA from the latest source of moveit. I guess the missing transmissions issues was in the deb package version of for ROS kinetic and ubuntu 16.04.
I have not got time to test it. It seems like I need to put together a launch file manually.

MohmadAyman, 2019-01-28:
@superxingzheng  you can also find a `demo_gazebo.launch` :+1: 

If that answers your question, would you please close the issue :+1:

closed 2019-01-28 by superxingzheng