moveit/moveit #1310: More effective way to setup Self-Collision Matrix

RhysMcK, 2019-01-17:
### Description

When dealing with complex urdf's which change frequently depending on application, it not viable to use the moveit setup assistance which will auto generate the self-collision matrix ( i.e <disable_collisions> tag in srdf). Hence, manually setting up all the disable collisions tags becomes quite cumbersome, especially with robots which have a lot of links. 

Would it be straight forward to implement an alternative options where all self collision are disabled by default and specific link pairs which need to be check for collisions can be enabled by a similar tag ( i.e <enable_collisions> tag in srdf for example)? 

Any thoughts on this?

Cheers.

rhaschke, 2019-01-17:
Usually, checking for collisions is the default and ignoring potential collisions is the exception. That's why the allowed collision matrix (ACM) only stores those exceptions.
Trying to initialize the matrix the other way around, would probably pose a performance issue as well.
Do you know about the `collisions_updater` executable? That's a command-line tool to generate the ACM.

RhysMcK, 2019-01-21:
Thanks for the response! No I haven't seen the collisions_updater executable before. Is there any documentation on how to use it? Can't seem to find anything about it.

jonbinney, 2019-01-21:
@RhysMcK I wasn't able to find documentation on the web for it, but in the code it looks like the command line args have descriptions: https://github.com/ros-planning/moveit/blob/melodic-devel/moveit_setup_assistant/src/collisions_updater.cpp

BryceStevenWilley, 2019-06-12:
@RhysMcK This seems to solve your case. Closing, but feel free to open it again if collisions_updater didn't work as well as you wanted.

closed 2019-06-12 by BryceStevenWilley