moveit/moveit #1311: rejected fetching tag 0.10.6 (would clobber existing tag)

v4hn, 2019-01-21:
I see this over here at the moment:

```
moveit $ git fetch --tags
From git://github.com/ros-planning/moveit
 ! [rejected]            0.10.6     -> 0.10.6  (would clobber existing tag)
```

Did anyone change the tags? @130s @rhaschke

130s, 2019-01-21:
I haven't done any release work since I made [0.10.5](https://github.com/ros-planning/moveit/releases/tag/0.10.5).

Can you provide steps of how to reproduce when it happens? 
(I couldn't reproduce by running `wstool update -t -v src` over my MoveIt! workspace.)

rhaschke, 2019-01-21:
I have force-pushed tags during the last release cycle(s) to only keep a single tag per release.
As we observed several release issues lately (e.g. with chomp stuff), it was a lengthy and incremental process to get MoveIt compiled on the ROS build farm. In any case, I only kept the final release that actual went through.
Looks like you pulled during this release process and thus have an outdated tag locally. Remove the local tag with `git tag -d 0.10.6` and fetch again.

closed 2019-01-23 by rhaschke