moveit/moveit #1313: how to handle tcp collision avoidance

nashdingsheng, 2019-01-22:
### Description
i want to consult  if there are some efficient ways to solve the collision produced by tcps on moveit.
Overview of your issue here.
         hi, i am developing motion planning module for surgical robot on ur5, and there are several TCPs(tool-center-point) for instance,calibration board,medical operation tools.  when i let the robot move to some target places and postures, it may probably collide with robot links, and i have to manually add waypoints to avoidance this collision, it is time consuming and definitely not practical in the real circumstance. it i do not know whether moveit! has the related part or not on tcp collision avoidance, if it has and how to solve .Ultimately i do not know whether there are some good examples or paper for me to refer to , thank you very much.

### Your environment
* ROS Distro: [Indigo|Jade|Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build? Binary
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

nashdingsheng, 2019-01-22:
sorry, i will ask question to [answers.ros.org](url) in the future. maybe i misunderstand the concept of tcp, let us forget tcp just physical tools collision avoidance i actually referred to . you mean the method to settle Real physical tools collision problem is to define its collision model in urdf file, and when planning, moveit will togetherly with other links  consider its collision avoidance? i am newer to it, and slowly to know the details about planning,thank you!

rhaschke, 2019-01-22:
Yes, if there is a collision model for your tool in the urdf, MoveIt will consider it during planning.

closed 2019-02-19 by v4hn