moveit/moveit #1315: FCL collision check does not return contacts

bhomberg, 2019-01-22:
### Description

When using the PlanningScene collision check, even when there are collisions, the response messages is not filled with the list of the contacting objects.    

### Your environment
* ROS Distro: Melodic
* OS Version: 18.04
* Source, 56745eb1b43d9bee1df8043481d9a08d22f54a83

### Steps to reproduce
The line of code in question that we're running is: planning_scene->checkCollision(req, res, *state, allowed_collision_matrix);

### Expected behaviour
I would expect that if res.collision is true, then res.contacts will contain the list of those collisions.

### Actual behaviour
res.contacts is empty

It looks like PlanningScene uses CollisionWorldFCL underneath; digging into https://github.com/ros-planning/moveit/blob/melodic-devel/moveit_core/collision_detection_fcl/src/collision_world_fcl.cpp, the checkRobotCollisionHelper function takes in the response message but does not seem to fill it.  Is that intended behavior?  Is there something preventing that information from being filled in?  I haven't dived deeper than this level this in debugging yet.

rhaschke, 2019-01-22:
The [`CollisionRequests`](https://github.com/ros-planning/moveit/blob/6929b42d1491c04c2f45fa1bd8d9c5a0a177cfcc/moveit_core/collision_detection/include/moveit/collision_detection/collision_common.h#L174-L221) structure has several fields to control the behaviour of collision checks.
Particularly, there is a [`boolean`, named `contacts`](https://github.com/ros-planning/moveit/blob/6929b42d1491c04c2f45fa1bd8d9c5a0a177cfcc/moveit_core/collision_detection/include/moveit/collision_detection/collision_common.h#L200-L201), which indicates that contacts are desired ;-)

closed 2019-01-22 by rhaschke

jonbinney, 2019-01-22:
@rhaschke i believe that @bhomberg is referring to the `CollisionResult`, not the request. The result is passed in by reference; presumably to be filled in with the result. In this case the detected collisions aren't filled in by FCL. Do you know if that is intended, or just an oversight?

rhaschke, 2019-01-22:
Exactly, you need to explicitly request to return contacts.