moveit/moveit #1317: Optimizing self-collision checks?

simonschmeisser, 2019-01-23:
After changing the FCL cache for world and attached objects (#1316) now self collision checking shows up as a remaining big consumer of time. I know that the combination of joint limits and Allowed Collision Matrix (ACM) allows to shortcut some collision checks here. However I still wonder if we could do more? Something like a "conditional" ACM maybe? (example: if j1 < 90°, don't check "left side-pane" at all)

Any ideas? Other approaches? (I ask mainly out of curiosity as I think I now need to optimize different parts of the system ... )

@Levi-Armstrong

Levi-Armstrong, 2019-01-25:
I think it would help but when monitoring the solving of the motion plan, moveit reconstruct the collision environment every time a collision check is requested. This is a large percentage when performing thousands of checks also not directly related to moveit current there is not a way to indicate if a collision object is a convex hull so everything is represented as a detailed mesh. Changing these two factors provides a 5-10x improvement.

Levi-Armstrong, 2019-01-25:
I am currently working making the collision package with tesseract_collision self contained. Once finished I will start discussion with the moveit maintainers on how they would like to approach integrating the new capability as it will require some major changes to the api to get the performance benefits.