moveit/moveit #1323: MoveIt!'s Test Utils Aren't Include-able

davetcoleman, 2019-01-24:
This is a continuation of a conversation with @rhaschke [here](https://github.com/ros-planning/moveit/pull/1272#discussion_r247324346):

> I actually tried to move these functions to libmoveit_test_utils, because I like the idea. However, this causes some serious issues with clang. Looks like gcc performs lazy linking and does not actually link against libs that are not required by the final binary. On the contrary, clang seems to link each and every library, no matter it is required or not. However, as libmoveit_test_utils finally requires some symbols from gtest, this fails. We cannot also simply link libmoveit_test_utils against gtest as gtest is not installed - and I think libmoveit_test_utils shouldn't be installed neither. I couldn't find any hints how to resolve this. 

This is an issue if we have a test library that can't be used in tests.

Why is libmoveit_test_utils linking against gtest? I looked at the .h and .cpp and there were no includes or mentions of gtest. Same in the CMakeList. What am I missing? Can we remove the dep on gtest?

davetcoleman, 2019-01-24:
@BryceStevenWilley thoughts?

BryceStevenWilley, 2019-01-24:
Sorry, I remember seeing the original thread, but haven't had the time to dive into what was happening yet. 

So from what I understand:
* At the moment, libmoveit_test_utils does *not* need gtest. It's just a utility for making robot models. The test moniker was because it wasn't very rigorious, i.e. the robot models made shouldn't be used for anything but simple unit tests.
* @rhaschke added some [nice helper functions](https://github.com/ros-planning/moveit/pull/1272/files#diff-39cfedc07d9eee25f617887ae9f041e1R241) specifically using gtest, and those are what we can't add to libmoveit_test_utils. We could take the alternate route here by simply using gtest-free versions of those helpers: returning `bool`s instead of `testing::AssertionResult`. Downsides to this are that the test code will be less readable, and less helpful information will be given when a test fails.
* As for the linking issue in gtest, rhaschke is correct when he says that 
> [gtest] shouldn't be installed. Actually, it's Google's philosophy that each and every software package builds gtest on its own from source.

After trying to understand this for a while, I don't think having a library (libmoveit_test_utils) that wraps around gtest, linking to it, while other projects link to libmoveit_test_utils, is a common use case of gtest. The 'correct' gtest way would be to include gtest in every project's source tree as [external code](https://github.com/google/googletest/tree/master/googletest#incorporating-into-an-existing-cmake-project).  

However, I don't quite understand the statement: "GTEST_LIBRARIES are not found from another package than moveit_core". Getting GTEST_LIBRARIES implies we're using [FindGtest](https://cmake.org/cmake/help/v3.0/module/FindGTest.html), but I don't understand enough of how this works to know why it works in moveit_core, but nowhere else.

davetcoleman, 2019-01-24:
Thanks, now I understand.

This is really too bad - it will result in a lot of testing code duplication in every MoveIt! package, and possibly inconsistent testing approaches in each package.