moveit/moveit #1326: Releasing MoveIt! 1.0

davetcoleman, 2019-01-25:
Hi all,

In case you don't follow MoveIt! Discourse (which you [should](http://moveit.ros.org/about/get_involved/)!) we are planning to [release MoveIt! 1.0 on March 1st](https://discourse.ros.org/t/versioning-roadmap-moveit-1-0-release-plan/7437). I believe our main effort will be on our wonderful and talented release manager @130s to update package.xml's and CHANGELOGS, as well as some updates to the website. 

Before the official announcement, I'd like to create the new ``master`` branch also, as described on Discourse. **It would be ideal if we could get as many old PRs merged in now as possible** so long as they aren't changing API. 

What else needs to be done to prepare for this exciting milestone for MoveIt!?

rhaschke, 2019-01-25:
After IROS deadline (March 1st + extension?) I can help on the release.

130s, 2019-02-15:
Just an ACK - I'll work on this on the weekend Feb 16, 17 and get back here.

v4hn, 2019-02-19:
As @rhaschke pointed out already, the deadline on March, 1st coincides with the IROS submission deadline.
I have been (and am) very busy before this deadline and have few resources for anything else (you probably noticed...). So I will sadly not get around to do much more before then.

davetcoleman, 2019-02-22:
I'm traveling next week so its actually bad for me also. Let's delay the release to Friday March 8th.

Note I updated the top post with a checklist of TODOs

130s, 2019-02-23:
I might appreciate clarification. Do we want to make 1.0.0 release:
- from `master` branch only?
  - For all ROS distro MoveIt! currently supports?
- from `master` branch, in addition to distro-based branches?
  - If so which distro bases off from `master` and which doesn't?

davetcoleman, 2019-02-24:
We only want to call Melodic 1.0, and i think its fine if we release from ``melodic-devel``. The branching for ``master`` should happen after we version increment to 1.0

130s, 2019-02-25:
> - [ ] Release to bloom @130s

Done https://github.com/ros/rosdistro/pull/20372

v4hn, 2019-02-26:
> > Let's delay the release to Friday March 8th.

> Done [ros/rosdistro#20372](https://github.com/ros/rosdistro/pull/20372)

@130s you released prematurely

@davetcoleman @rhaschke @bmagyar and others
Can we live with that or is there API changes that should be merged before?

> https://github.com/ros/rosdistro/pull/20372

Personally and without looking through things, I don't know of anything specific,
but still we probably want to schedule a 1.1 release in two weeks if there are bigger changes coming in until then.

rhaschke, 2019-02-26:
I don't feel so comfortable with the early release too. Chomp has several segfault issues that should be fixed before a release. Also, currently [MSA fails on the ROS build farm](http://build.ros.org/view/Mbin_uB64/job/Mbin_uB64__moveit_setup_assistant__ubuntu_bionic_amd64__binary/). As I said, I don't have time to help before the IROS deadline.

130s, 2019-02-26:
Ok. Knowing there's less chance for the fix, I'll just revert rosdistro entry.

I'm sorry, I should've communicated better. It's just what "release" refers to is unclear. Since "running bloom" was at the top of the list of action items, I thought we were ready for bloom.

130s, 2019-02-26:
Revert request https://github.com/ros/rosdistro/pull/20391

v4hn, 2019-02-26:
In my understanding "release" always refers to uploading a version tag to the public git repository.
So, in my understanding we still released and should bump the version number again later.

Thanks for the revert!

rhaschke, 2019-02-26:
> In my understanding "release" always refers to uploading a version tag to the public git repository.

I don't share this understanding :wink: For me, a release is published, when it actually reached `ros-shadow-fixed`.

davetcoleman, 2019-02-27:
Bumping the patch version seems pretty harmless, to make clear the previous early release mistake. But I don't have a strong opinion

bmagyar, 2019-02-27:
It feels a little hacky but I'd try to make the first release be 1.0.0 even if it takes changing the existing tag in git... 

As affected by IROS as the others, I'll put some effort into the seed trajectory PR.

@rhaschke is there a list of known segfaults on an issue? If not, should we start creating one?

rhaschke, 2019-02-27:
@bmagyar I didn't filed an issue yet and last time I wanted to look into the segfault issue, I wasn't able to reproduce it easily.

davetcoleman, 2019-02-28:
From today's maintainer meeting, I plan on creating the master branch in the next day. This will allow a "code freeze" period for the ``melodic-devel`` branch for the next week before we make the 1.0 release

130s, 2019-03-04:
From my perspective as a release handy-person, only blocker for making a patch release (1.0.1) is #1371. Is there any other issues/PRs that need to be addressed before another release? If there's none, I'll plan to run bloom on March 6th.

130s, 2019-03-04:
https://github.com/ros-planning/moveit/issues/1371#issuecomment-469391005 sounds like the only way we can "test" if #1371 is fixed or not is to bloom and see if the issue happens on buildfarm. Though that's far from ideal, if we're ok with it, I can bloom to include #1373 and see.

rhaschke, 2019-03-04:
I'm pretty confident that this should fix. But, yes, I'm afraid blooming is the only way to prove it.
However, please wait before the next release until March, 8th as agreed in https://github.com/ros-planning/moveit/issues/1326#issuecomment-466588834.

davetcoleman, 2019-03-04:
I don't think it hurts to run it on Wed instead if we aren't sure it's going to take. I'd like to do the announcement on Friday that it works.

130s, 2019-03-05:
Ok. I can run bloom even tonight or tomorrow (Tues) if we just want to see if https://github.com/ros-planning/moveit/issues/1371 is fixed. @rhaschke Would you agree or should we still wait?

rhaschke, 2019-03-05:
We still have some open issues that should be resolved before the next release attempt:
* [ ] #1377

davetcoleman, 2019-03-05:
Thanks for working on that fix. However, CHOMP is a problematic planner already, and not at 100% stability. I don't think it should block the 1.0 release, its never worked well in recent years... we can fix it in a future release. If this was OMPL that had the issue, then we should pull the brakes. 

Also the point of running bloom tonight (Tue) is to make sure we addressed the MSA regression. Given that we already delayed the release by a week, we need to verify this stuff now. @130s we need to charge ahead.

davetcoleman, 2019-03-06:
The [new website update](https://github.com/ros-planning/moveit.ros.org/pull/261), blog post, and tutorial updates ready to go, just pending actual bloom release.

mlautman, 2019-03-08:
@rhaschke Do you need help on any of your PR's before the release tomorrow? I have some bandwidth today and tomorrow

rhaschke, 2019-03-08:
@mlautman, I think everything except #1382 is prepared. #1382 already has a positive review from Henning. As this is a critical code section and @davetcoleman get's nervous looking at it, it would be nice to have just another review for it.

davetcoleman, 2019-03-08:
I've merged it after getting an additional review.

Let's release 1.0!

130s, 2019-03-08:
Getting stuck with bloom. It wants to release 1.0.0-1, instead of 1.0.1. I keep looking into it.

```
$ bloom-release --rosdistro melodic --track melodic moveit
:
Executing release track 'melodic'
==> bloom-export-upstream /tmp/tmpDJMLu9/upstream git --tag 1.0.0 --display-uri https://github.com/ros-planning/moveit.git --name upstream --output-dir /tmp/tmp4FH_Xf
Checking out repository at 'https://github.com/ros-planning/moveit.git' to reference '1.0.0'.
Exporting to archive: '/tmp/tmp4FH_Xf/upstream-1.0.0.tar.gz'

Cloning into '/tmp/tmpn3Z6Ya'...
warning: --depth is ignored in local clones; use file:// instead.
done.                                                            
md5: b923cec81af927a92a4055152b70a447                            
==> git-bloom-import-upstream /tmp/tmp4FH_Xf/upstream-1.0.0.tar.gz  --release-version 1.0.0 --replace
The latest upstream tag in the release repository is 'upstream/1.0.0'.                               
Removing tag: 'upstream/1.0.0'                                                                       
Importing archive into upstream branch...                                                            
Creating tag: 'upstream/1.0.0'                                                                       
I'm happy.  You should be too.
:
```

130s, 2019-03-08:
Resolved. Now bloom is running.

I had to update the release repository config with `bloom-release --edit`, to point to `master` branch (it was set to `melodic-devel`).
```
$ bloom-release --rosdistro melodic --track melodic moveit --edit
:
Upstream Devel Branch:
  <vcs reference>     
    Branch in upstream repository on which to search for the version.
    This is used only when version is set to ':{auto}'.              
  ['melodic-devel']: master
:
Executing release track 'melodic'
==> bloom-export-upstream /tmp/tmpQUBYUs/upstream git --tag 1.0.1 --display-uri https://github.com/ros-planning/moveit.git --name upstream --output-dir /tmp/tmpPadY8m
Checking out repository at 'https://github.com/ros-planning/moveit.git' to reference '1.0.1'.
Exporting to archive: '/tmp/tmpPadY8m/upstream-1.0.1.tar.gz'
```

130s, 2019-03-08:
> - 1.0.1
>   -  bloom

Done https://github.com/ros/rosdistro/pull/20516

rhaschke, 2019-03-08:
Dear @davetcoleman,

as you cherry-picked #1382 from master to melodic-devel, both branches have now diverged. 
The 1.0.1 release was made from master. As melodic-devel should become the "release branch" for 1.0.x releases, I think melodic-devel should be reset to 1.0.1.

@130s, thanks for releasing. Hopefully the ROS buildfarm succeeds!

davetcoleman, 2019-03-08:
@rhaschke I'm confused, I thought I was suppose to cherry-pick to melodic.

> both branches have now diverged.

What do you mean by this? By cherry-picking, it keeps them in sync.

> The 1.0.1 release was made from master

Why was this the case? The ``melodic-devel`` branch was suppose to be the 'code freeze' branch as the candidate for 1.0. Was something miscommunication here?

rhaschke, 2019-03-08:
> > both branches have now diverged.

> What do you mean by this? By cherry-picking, it keeps them in sync.

They are in sync content-wise, but obviously, the diverged in git.

> > The 1.0.1 release was made from master

> Why was this the case? The melodic-devel branch was supposed to be the branch for 1.0. Was something miscommunication here?

Obviously, @130s didn't noticed that.

davetcoleman, 2019-03-08:
Our plan is to target almost all development in ``master``, so isn't this divergence inevitable? Still not clear to me what I should have done differently. Cherry-picking has always been our policy between branches.

rhaschke, 2019-03-09:
> Still not clear to me what I should have done differently.

I will try to explain in more detail below.

> Our plan is to target almost all development in master, so isn't this divergence inevitable?

In the future both branches will diverge, I agree. However, right now, they should be in sync.
The master branch exists for at least a week now and we have merged several PRs targeting either master or melodic-devel. However, I always fast-forwarded the other branch correspondingly to keep both branches in sync. There was no value in cherry-picking, which would have meant that the same changesets were applied in parallel, twice, to both branches. So for #1382, you should have fast-forwarded melodic-devel as well.

> Cherry-picking has always been our policy between branches.

Cherry-picking only makes sense, when you want to _selectively_ pick individual commits from one branch. If you essentially want to keep all commits of branch A also in branch B, branch A should be simply merged into branch B from time to time. That's also the procedure we follow for MoveIt 2: The melodic-devel branch of MoveIt 1 is regularly merged into the Moveit 2 master.

The same procedure I suggest for melodic-devel and master in MoveIt 1:
- Bug fix commits, that should be part of both, `melodic-devel` and `master` should be initially targeted and merged into `melodic-devel`.
- New feature and ABI-breaking changes should be committed to `master` only.
- `melodic-devel` commits are regularly merged into `master`.

The resulting commit history clearly captures which commits are only available in the master branch as you can see from the following example, where the black branch follows the commits on the red one:
[image]

davetcoleman, 2019-03-19:
Thanks for the explanation @rhaschke 

I'm closing this issue as moveit 1.0 is released. Thanks again @130s !

closed 2019-03-19 by davetcoleman