moveit/moveit #1329: OMPL plugin: simplifyMax if planner exceeded time limit

simonschmeisser, 2019-01-29:
I currently try to optimize canceling/terminating OMPL based planners. While looking at the code I noticed a strange 'feature'. In `ModelBasedPlanningContext`
https://github.com/ros-planning/moveit/blob/405557df02a06e19dd4dcfa0ee4a0a8472fd34ef/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L573

the used time is subtracted from the allowed time. This could lead to a negative result (or theoretically) zero. This is then passed to `ompl::geometric::SimpleSetup::simplifySolution` where `t < epsilon` is handled by `simplifyMax`, ie simplify as long as possible. 
https://github.com/ompl/ompl/blob/d39f0f64418cf047de173c757a0c1e70661e3740/src/ompl/geometric/src/SimpleSetup.cpp#L169

Also passing in a max time instead of a `ompl::base::PlannerTerminationCondition` means we cannot actually cancel/terminate a Simplification.

I think we should actually have a) a separate config parameter for simplification time  and b) be able to terminate simplification

Opinions?

simonschmeisser, 2019-01-29:
It's quite simple:
```
void ompl_interface::ModelBasedPlanningContext::simplifySolution(double timeout)
{
  ob::PlannerTerminationCondition ptc = ob::timedPlannerTerminationCondition(timeout);
  registerTerminationCondition(ptc);
  ompl_simple_setup_->simplifySolution(ptc);
  unregisterTerminationCondition();
  last_simplify_time_ = ompl_simple_setup_->getLastSimplificationTime();
}
```

but terminating simplification (as it currently is) does not seem to help too much as the `atLeastOnce` parameter will still make it run once (and keep running which takes 700ms to more than a second in some cases, depending on longest_valid_segment_fraction and the complexity of the solution)

simonschmeisser, 2019-01-29:
Also if a planner gets canceled/terminated while already having a valid solution, we still simplify it

zkingston, 2019-01-29:
We can get better control of the simplification process by grabbing the `ompl::geometric::PathSimplifier` from the `SimpleSetup` (see [here](https://github.com/ompl/ompl/blob/d39f0f64418cf047de173c757a0c1e70661e3740/src/ompl/geometric/SimpleSetup.h#L123)).

Something like the following:
```cpp
void ompl_interface::ModelBasedPlanningContext::simplifySolution(double timeout, bool atLeastOnce)
{
  ob::PlannerTerminationCondition ptc = ob::timedPlannerTerminationCondition(timeout);
  auto &psk = ompl_simple_setup_->getPathSimplifier();
  const auto &pdef = ompl_simple_setup_->getProblemDefinition();
  
  // Check to make sure the solution is something we actually want to simplify.
  if (!pdef->hasExactSolution())
    return;

  const base::PathPtr &p = pdef->getSolutionPath();

  ompl::time::point start = ompl::time::now();
  
  // Simplify using timeout and our atLeastOnce
  psk->simplify(static_cast<PathGeometric &>(*p), ptc, atLeastOnce);

  last_simplify_time_ = ompl::time::seconds(ompl::time::now() - start);
}
```