moveit/moveit #1330: can't get padding to work!

skasewa, 2019-01-29:
### Description

I'm just trying to add padding to my robot model so that it reports collisions around some margin of the robot when doing a collision check, but I just cannot seem to figure out how to increase the padding on the robot model.

From what I can tell, no matter what I do, the padding remains 0.0

Following other comments here, I have added a padding.yaml and put in various numbers for the parameters (including very large numbers) but they have no effect whatsoever.

```
default_robot_padding: 1.0
default_robot_scale: 1.0
default_object_padding: 1.0
default_attached_padding: 1.0
```

I am not the first one to report this behaviour (I have seen other issues reporting similar things from 2014).

Could someone help me with:
- a working example where the padding value actually changes the behaviour of collision checking and planning
- a full working example with PlanningSceneMonitor instead of PlanningScene (given that apparently one isn't supposed to use PlanningScene, but there are no samples/tutorials provided that do not use PlanningScene)

I'm trying to figure out where the padding values eventually affect the robot model. It's quite tricky to trace through what goes on once a checkCollision call is made. It summons CollosionRobot through `active_collision_->getCollisionRobot();` in line 203 of `planning_scene.h ` but I cannot seem to figure out what active_collision_ is pointing to, to trace it further back up the chain. My sense is that the getCollisionRobot() does not pad the robot model, or whatever padding values it accesses are one of the several default 0.0 values.

I'm surprised this isn't better investigated or documented, but maybe I have just not been able to find suitable sources. Any directions addressing my concerns would be deeply appreciated; I have been struggling to get this to work for over a day now.

* ROS Distro: Kinetic
* OS Version:  Ubuntu 16.04
* Binary

### Steps to reproduce
* Load any robot model with some padding
* Add any collision object in the scene (not currently colliding with the robot)
* Run collisionCheck (It will report false)
* Repeat with increasing pad values
* Notice that no matter what the pad values, the collisionCheck is always false

### Expected behaviour
* If the object collides with the padded robot, collisionCheck should report it
* It might be that I am just implementing something wrong (e.g by using padding.yaml + rosparam in launch file), so I would be grateful if you could upload a sample which works.

### Actual behaviour
* Repeat with increasing pad values
* Notice that no matter what the pad values, the collisionCheck is always false

### Backtrace or Console output
Not provided (I may carve out a minimum example to put up later, but I have already spent too long on this)

simonschmeisser, 2019-02-03:
Please have a look at the link to the code in issue https://github.com/ros-planning/moveit/issues/1116

I guess that you need to put those parameters somewhere else ... You can add some ROS_ERROR debug messages to verify that you actually put the parameters in the correct spot on the parameter server.

Once that works do note some limitations of the padding: it assumes convex bodies: https://github.com/ros-planning/geometric_shapes/issues/76
and you cannot configure robot self-collision padding https://github.com/ros-planning/moveit/issues/683

I'll be back in office on Thursday and might find an example somewhere but I did not look into it too much yet

FSund, 2021-03-25:
Since this is one of the top results on Google for "moveit default_robot_padding" I thought I could leave a working example here after spending a couple hours figuring out how to correctly set `default_robot_padding`.

Add a file `padding.yaml`, for example in `<robot>_moveit_config/config`, with the following content

```
# parameters for padded robot (not used for self collision checks)
default_robot_padding: 0.05
default_robot_scale: 1.0
```

and load the parameters from `padding.yaml` into the correct namespace in a launch file somewhere, for example in `planning_context.launch`

```
<!-- Load padding -->
<group ns="move_group">
  <group ns="$(arg robot_description)_planning">
    <rosparam command="load" file="$(find <robot>_moveit_config)/config/padding.yaml" />
  </group>
</group>
```

So it goes in the `/move_group/robot_description_planning/` namespace.

I have verified that this works as expected on my machine (Ubuntu 20.04, with ROS Noetic). Setting a padding of for example 0.1 makes the planner report a collision when I move it about 10 cm from an obstacle in the scene in RViz.

mintar, 2021-05-31:
Just a further note for everyone else who stumbles upon this: I'm using a prefix (`mobipick`) for my robot. It used to work exactly how @FSund described:

`mobipick/move_group/robot_description_planning/default_attached_padding: 0.0`

In the current ROS Noetic version of MoveIt (1.1.5), there's a second `mobipick` necessary to make it work:

`mobipick/move_group/mobipick/robot_description_planning/default_attached_padding: 0.0`

v4hn, 2021-05-31:
@mintar This is definitely a bug. Can you pinpoint the offending commit/faulty logic (and maybe file a separate issue)?

v4hn, 2021-05-31:
I don't think there is an issue anymore with the original report. I created #2699 for the issue mintar just raised.

closed 2021-05-31 by v4hn

mintar, 2021-06-02:
There's now a PR to fix this: #2706

Once the PR is merged, the new param path will be the regular `/robot_description_planning/` (or `/<namespace>/robot_rescription_planning` if you're using a namespace) like all other PSM parameters. So instead of this:

```
<!-- Load padding -->
<group ns="move_group">
  <group ns="$(arg robot_description)_planning">
    <rosparam command="load" file="$(find <robot>_moveit_config)/config/padding.yaml" />
  </group>
</group>
```

... you simply have to do this:

```
<!-- Load padding -->
<group ns="$(arg robot_description)_planning">
  <rosparam command="load" file="$(find <robot>_moveit_config)/config/padding.yaml" />
</group>
```