moveit/moveit #1331: Problem in building MoveIt! from source

pulkit123tak, 2019-01-31:
### Description

Hi all, I am building MoveIt! from source from this link :[https://github.com/ros-planning/moveit/tree/kinetic-devel]

### My environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04

After cloning the entire package from the mentioned repository I used catkin build to build it. However the output shows these errors: 

> Errors     << moveit_ros_visualization:cmake /home/bhoot/effort_ws/logs/moveit_ros_visualization/build.cmake.002.log
> CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt5Core/Qt5CoreConfig.cmake:27 (message):
>   The imported target "Qt5::Core" references the file
> 
>      "/usr/lib/x86_64-linux-gnu/qt5//mkspecs/linux-g++-64"
> 
>   but this file does not exist.  Possible reasons include:
> 
>   * The file was deleted, renamed, or moved to another location.
> 
>   * An install or uninstall procedure did not complete successfully.
> 
>   * The installation package was faulty and contained
> 
>      "/usr/lib/x86_64-linux-gnu/cmake/Qt5Core/Qt5CoreConfigExtras.cmake"
> 
>   but not all the files it references.
> 
> Call Stack (most recent call first):
>   /usr/lib/x86_64-linux-gnu/cmake/Qt5Core/Qt5CoreConfigExtras.cmake:50 (_qt5_Core_check_file_exists)
>   /usr/lib/x86_64-linux-gnu/cmake/Qt5Core/Qt5CoreConfig.cmake:158 (include)
>   /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:26 (find_package)
>   CMakeLists.txt:54 (find_package)
> 
> 
> cd /home/bhoot/effort_ws/build/moveit_ros_visualization; catkin build --get-env moveit_ros_visualization | catkin env -si  /usr/bin/cmake /home/bhoot/effort_ws/src/moveit/moveit_ros/visualization --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/bhoot/effort_ws/devel/.private/moveit_ros_visualization -DCMAKE_INSTALL_PREFIX=/home/bhoot/effort_ws/install -DCMAKE_BUILD_TYPE=Release; cd -
> 

I am unable to find out what is the problem. Is it due to my system or through MoveIt! ??

rhaschke, 2019-01-31:
Looks like you broke your Qt5 installation. Try to reinstall Qt5.

pulkit123tak, 2019-01-31:
I have reinstalled Qt5 from this [link](https://wiki.qt.io/Install_Qt_5_on_Ubuntu) , but the error is still coming up.

rhaschke, 2019-01-31:
Why didn't you simply reinstall the corresponding debian packages?
Install from the external Qt5 sources installs into a different location and doesn't help to fix your problem.

closed 2019-02-02 by rhaschke