moveit/moveit #1336: Baxter planning joints constraints

Enrichetto, 2019-02-01:
### Description
Hello,I'm working with baxter and i'm using ros and movit in order to manipulate an articulated object. Now i want add some contraint on the joints of the baxter's arms in order to reach in better way the object. Let's say that i want, for example, that the joint of baxter "right_lower_shoulder" rotate just inside of some limited range.
/*i add this part. the values are totally random but in any case i think that this approach is not correct.*/
geometry_msgs::PoseStamped enrico;std::vector<double> tolerance_pose(3, 100);std::vector<double> tolerance_angle(3, 100);
enrico.pose.position.x = 0.32;
enrico.pose.position.y = -0.25;
enrico.pose.position.z = 0.65;
enrico.pose.orientation.w = 1.0;
moveit_msgs::Constraints pose_goal_2 =    kinematic_constraints::constructGoalConstraints("right_lower_shoulder", enrico, tolerance_pose,tolerance_angle);
/*until here
 root_action_descriptor.gripper_pose.pose.position.z -= 0.11;         child_action_descriptor.gripper_pose.pose.position.z -= 0.11;
// move_group.setPathConstraints(pose_goal_2) is what i add in order to have a constraint
move_group.setPathConstraints(pose_goal_2);
move_group.clearPoseTargets();
move_group.setPoseTarget(root_action_descriptor.gripper_pose, root_action_descriptor.gripper_frame);
move_group.setPoseTarget(child_action_descriptor.gripper_pose, child_action_descriptor.gripper_frame);
move_group.setStartStateToCurrentState();
if (!move_group.move())
{            ROS_ERROR("[PACO increase action]: unable to reach target positions. 2");
            exit();
            return false;        }

This code come from :https://github.com/EmaroLab/paco_actionsThank you for you help!

jonbinney, 2019-02-01:
@Enrichetto this seems like it would be better to ask on ROS Answers. Can you ask it there? Then if it turns out that there is a bug in moveit we can open an issue to fix it here.

rhaschke, 2019-02-08:
@Enrichetto I hope you got some feedback on ROS answers. The description here, is missing a clear question. Closing.

closed 2019-02-08 by rhaschke