moveit/moveit #1337: Fake controller manager: ability to dynamically reconfigure joint positions

Achllle, 2019-02-01:
Currently you can specify an initial configuration when using the `fake_controller_manager` through a list of (group, pose) tuples (see [tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/fake_controller_manager/fake_controller_manager_tutorial.html#yaml-file-examples)).

This only works on launch and requires specifying a list of configurations in advance. For testing purposes, it would be useful if we could change the configuration during runtime using a `sensor_msgs::JointState` message. You'd only need to launch one rostest and have multiple tests that initialize the configuration, potentially to a random set.

I'd be happy to work on this if this seems useful to others.

rhaschke, 2019-02-01:
The key of the `fake_controller_manager` is, that there is no real robot publishing it's joint states. 
Instead, the `fake_controller_manager` publishes those joint states when a plan is executed.
In order to move the robot to another pose, you can already now send a `JointState` msg to the `/joint_states` topic.
If you run any MSA-generated demo.launch, there is a parameter `use_gui:=true`, which will spawn a gui to control you robot's joints via sliders, exploiting exactly the described method.

closed 2019-02-01 by rhaschke

Achllle, 2019-02-01:
:man_facepalming: I knew this