moveit/moveit #1342: Link Travis artifacts for code coverage analysis and abi-compliance-checker

rhaschke, 2019-02-10:
It would be great if we could link some build artifacts of Travis CI from github:
- [CodeCov](https://docs.codecov.io/docs) is an interesting github plugin to visualize code coverage's analysis results.
- `abi-compliance-checker` results (in form of .html files) would need to be uploaded from Travis to some storage place and linked, either from Travis or from [github](https://github.com/travis-ci/travis-ci/issues/9828).

I think CodeCov uses its own storage at https://codecov.io.
For abi-compliance-checker we would need some external storage place to push those htmls too.
@davetcoleman Do you have some space on a PickNik server for this? I guess we should also be able to setup a cron-job to delete old files automatically.

Would be great if somebody could look into https://codecov.io.

davetcoleman, 2019-02-13:
Yea we have an SSH or FTP server I'm happy to provide for moveit, what type of file upload does it need?

rhaschke, 2019-02-14:
Regarding ABI check results, for download a normal http server would be required. For upload anything available from Travis' bash should be fine. However, I just noticed that [PR builds cannot access encrypted environment variables](https://docs.travis-ci.com/user/encryption-keys), which would be needed to secure access to your server.
Maybe instead of using a password, you could restrict server upload by file type (html), source IP address range (Travis), and file size?

To cache ABI check dumps (of the latest release), we could employ [Travis' caching mechanism](https://docs.travis-ci.com/user/caching/#fetching-and-storing-caches).

v4hn, 2019-02-21:
Previously I argued against ABI checks in CI,
because they take quite a long time to complete (compiling everything with -Og) and are not straight-forward to interpret.
Among other things, they often include false-positive ABI changes because of minor changes in the use of boost-datastructures.

That being said, I used an integrated offline script to verify ABI between indigo releases when we still did them: https://github.com/v4hn/moveit_abi_check

rhaschke, 2019-02-21:
I didn't run the ABI check for MoveIt on Travis yet, but I did for rviz. Yes, it takes some time...