moveit/moveit #1344: Weird invalid goal state

jingxixu, 2019-02-12:
### Description

I am using m1n6s200 to grasp a block. MoveIt! reports "Unable to sample any valid states for goal tree" when I try to manually set a goal using the interactive marker which is shown in the following screenshots. There is clearly no collision but I do not understand why the goal is invalid. I have spent many days on this and cannot figure out why. A good grasp needs to be very close to the target so that I have to move the arm there but MoveIt! is preventing me from doing that. Can you guys help? Thanks!

[image]
[image]
[image]

### Your environment
* ROS Distro: Kinetic
* OS Version: e.g. Ubuntu 16.04

jonbinney, 2019-02-12:
What area you launching? Are you writing your own code to call moveit C++ libraries, or are you calling move_group's action interface? Do you know what links are in collision?

felixvd, 2019-03-20:
@jingxixu Have you solved this problem?