moveit/moveit #1345: Moveit plan execution problem on a real robot via Rviz

Vonstafenberg, 2019-02-12:
### Description
Hi all,

I had multiple runs of Moveit without 'this' issue in the past. I have a UR5 arm on a mobile base. The moveit configuration for the whole platform was running smooth, both from Moveit Rviz and the c++ code. Recently I executed move it  and I am facing an issue now. I am having a feeling this is related to Moveit Rviz possibly due to consistent updates in Moveit. 
### Overview of Issue
Running Moveit on a simulated platform in Gazebo gives no issue. Running it on a Real robot produces this error. I referenced some similar issues like [this](https://github.com/wilselby/ROS_quadrotor_simulator/issues/22), and [this](https://answers.ros.org/question/257874/moveit-cannot-update-robots-state/) and [ref](https://answers.ros.org/question/254927/failed-to-validate-trajectory-after-updating-moveit-feb-2017/), [ref](https://answers.ros.org/question/252114/failed-to-validate-trajectory-couldnt-receive-full-current-joint-state-within-1s-error/?answer=276159#post-id-276159), but I since I am not an expert with Moveit and ROS, I have not come up with a conclusion to do a reasonable fix my self.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04.5
* Source or Binary build? : Binary

### Steps to reproduce
I don't think it is required but ....can provide Moveit config package.

### Expected behaviour
The plan computed via Moveit Rviz should execute.

### Actual behaviour
I get the warning below when using the **update** Goal or Current state in Moveit Rviz.
```
Maybe failed to update robot state, time diff: 3626,548s
``` 
and finally if  plan and then **execute**
> **ABORTED: Solution found but controller failed during execution**

### Backtrace or Console output

Here is the Console Ouput to provide more detail

**UR5 Drivers Launch**

```
started roslaunch server http://192.168.0.201:50815/

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /ur_driver/base_frame: arm_base
 * /ur_driver/max_payload: 5.0
 * /ur_driver/max_velocity: 1.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix: arm_
 * /ur_driver/robot_ip_address: 192.168.0.207
 * /ur_driver/robot_reverse_port: 50002
 * /ur_driver/servoj_time: 0.008
 * /ur_driver/tool_frame: arm_tool0_controller
 * /ur_driver/use_ros_control: False

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)

ROS_MASTER_URI=http://summitB:11311

core service [/rosout] found
process[robot_state_publisher-1]: started with pid [6193]
process[ur_driver-2]: started with pid [6194]
[ INFO] [1549987992.438155020]: Setting prefix to arm_

```

**Moveit Launch**

```started roslaunch server http://192.168.0.3:36135/

SUMMARY

PARAMETERS
[... 145 lines cut ...]
[ INFO] [1549991782.669495948]: Using solve type Manipulation1
[ INFO] [1549991783.156731702]: Starting scene monitor
[ INFO] [1549991783.210734866]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1549991784.150344676]: Constructing new MoveGroup connection for group 'arm' in namespace ''
[ INFO] [1549991785.963263956]: Ready to take commands for planning group arm.
[ INFO] [1549991785.963381751]: Looking around: no
[ INFO] [1549991785.963402961]: Replanning: no
[ WARN] [1549991785.980275288]: Interactive marker 'EE:goal_gripper_link' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.
```

**Warning on Updating Current or Goal state Via Moveit Rviz**
```
[ WARN] [1549991822.951927605]: Maybe failed to update robot state, time diff: 3626,548s
[ WARN] [1549991826.885332330]: Maybe failed to update robot state, time diff: 3626,738s
```

**Error Output on Plan Execute**
```
[ INFO] [1549991866.459670062]: ABORTED: Solution found but controller failed during execution
[ WARN] [1549991867.080251238]: Maybe failed to update robot state, time diff: 3626,666s
```

@gavanderhoorn and @v4hn  would you be kind to look into the issue

regards

jonbinney, 2019-02-14:
At first glance, that error rounds like it is saying that the UR5 controller is failling, not moveit? Can you rostopic echo the FollowJointTrajectory goal topic to make sure moveit is publishing it correctly? And then use `rosnode info` on the UR driver node to make sure it is listening on that topic?

Vonstafenberg, 2019-02-15:
@jonbinney ...Will sure check ...thank you for the feedback.

felixvd, 2019-05-20:
What was the result of your check?

RoBoLog, 2019-06-18:
@felixvd 

Hi,

thank you for asking, like jonbinney suggested, I did an echo on the " FollowJointTrajectory goal topic to make sure moveit is publishing it correctly ".... and Moveit was doing it correctly. So I updated the UR driver package from the repository. But still I have not been able to solve the problem.

felixvd, 2019-06-18:
The topics are matched, so the UR driver is listening to the topic that the goal topic is published to? Are there any errors when you start up the robot? Can you ping it? Are the IP settings etc. correct?

gavanderhoorn, 2019-06-18:
The error message shows this:

```
Maybe failed to update robot state, time diff: 3626,548s
```

that's suspicuously close to 3600 seconds, which would be an hour.

If there are multiple machines involved here, it could be that their clocks are not synced, or they are not in the same timezone / not set to the same timezone.