moveit/moveit #1346: Issue compiling MoveIt on ROS Melodic over Ubuntu 18.04

igorrecioh, 2019-02-13:
### Description

Not able to compile Moveit on ROS Melodic

### Your environment
* ROS Distro: [Melodic]
* OS Version:  Ubuntu 18.04
* Source or Binary build? Source
* If source, which git commit or tag? Melodic-devel

### Steps to reproduce
Trying to compile in a clean ws Moveit with  catkin_make_isolated --install 

### Expected behaviour
Compilation should complete successfully

### Actual behaviour
Fails in package moveit_core

### Backtrace or Console output
```
/home/erle/Desktop/igor/devel_isolated/moveit_core/lib/libmoveit_test_utils.so.0.10.8: undefined reference to `boost::re_detail_106501::cpp_regex_traits_implementation<char>::transform(char const*, char const*) const'
/home/erle/Desktop/igor/devel_isolated/moveit_core/lib/libmoveit_test_utils.so.0.10.8: undefined reference to `boost::re_detail_106501::cpp_regex_traits_implementation<char>::transform_primary(char const*, char const*) const'
collect2: error: ld returned 1 exit status
robot_state/CMakeFiles/robot_state_benchmark.dir/build.make:220: recipe for target '/home/erle/Desktop/igor/devel_isolated/moveit_core/lib/moveit_core/robot_state_benchmark' failed
make[2]: *** [/home/erle/Desktop/igor/devel_isolated/moveit_core/lib/moveit_core/robot_state_benchmark] Error 1
CMakeFiles/Makefile2:4123: recipe for target 'robot_state/CMakeFiles/robot_state_benchmark.dir/all' failed
make[1]: *** [robot_state/CMakeFiles/robot_state_benchmark.dir/all] Error 2
```

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2019-02-13:
Thanks for reporting this issue, but I think it's an issue on your system. We have CI jobs building with [`catkin tools`](https://travis-ci.org/ros-planning/moveit/branches) and with [`catkin_make_isolated`](http://build.ros.org/job/Mdev__moveit__ubuntu_bionic_amd64/) and both of them succeed.
Could you please try to build in a clean docker container, e.g. [`ros:melodic-ros-core-bionic`](https://github.com/osrf/docker_images/blob/eaca344ae304c30254451da89bae328eb65ee385/ros/melodic/ubuntu/bionic/ros-core/Dockerfile)?

igorrecioh, 2019-02-13:
Thanks for your quick reply, @rhaschke! We are trying to build it in a clean docker container as suggested. I did not mention before but I got MoveIt! code from [here](https://github.com/ros-planning/moveit). 

I will update this ASAP!

igorrecioh, 2019-02-13:
Different versions of Boost installed was the problem. Thanks for your help, @rhaschke !

closed 2019-02-13 by igorrecioh

rhaschke, 2019-02-13:
You are welcome.