moveit/moveit #1349: Realtime-ness of the OMPL planners

moucrob, 2019-02-15:
Assume one calls a given one of your planners by that [`bool ompl_interface::ModelBasedPlanningContext::solve(planning_interface::MotionPlanDetailedResponse& res)`](https://github.com/ros-planning/moveit/blob/32731d78be86b82e4445cd7222081b37b793c365/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L599) function, in order to get a trajectory of robot states as answer, and few other things. Then in order to process further, one has to wait until a boolean is returned depending on the success of the computation.
So if one indeed had a (real-time, precise) processing time constraint, he would time from right before to right after the line calling `solve`.
In my case, I first thought the whole `solve` process wasn't violating the `request_.allowed_planning_time`, that I believe is through what most of the users input their needs. Which is wrong now that I dove in its code. 

- I can see that the `interpolateSolution()` internal post-process is imposed whereas it doesn't have an authoritarian switch/countdown. 

- Furthermore, in case `simplify_solutions_` is set to true, the timeout inputed in `simplifySolution()` doesn't take in account the few `push_back`, `resize`, `reset` preceding its call.

Should I definitively deal with using as recorded time the `res.processing_time_`, or do you guys plan to modify those parts of the code sooner or later?

moucrob, 2019-02-15:
@gavanderhoorn Thank you, do you know by any chance a real-time motion planning library integrated into ROS ? Regards

gavanderhoorn, 2019-02-15:
This is going off-topic, but please specify what you mean by "real-time".

moucrob, 2019-02-15:
I mean some abstract `solve(`) function for calling any planner algorithm that would take in input a countdown constraint and for which if I record how long take the solve before returning true, the recorded time would be "almost" the same as the countdown constraint I inputed.

moucrob, 2019-02-15:
Currently I do am controlling the accuracy of the solving time by bounding the call with a time recorder, but I then am doing it at a "moveit layer level", and was just suggesting a warning in case people believe solve is real time and don't control the amount of time, or an improvement to handle it internally at an ompl-layer level, which would enable all their planners to be called real-time ones...

gavanderhoorn, 2019-02-15:
> I mean some abstract `solve(`) function for calling any planner algorithm that would take in input a countdown constraint and for which if I record how long take the solve before returning true/false, the recorded time would be "almost" the same as the countdown constraint I inputed.

I'm rather confused, as I asked whether you could clarify what you mean by the word "real-time". Should I understand "countdown constraint" to be a deadline?

"almost the same" is not really vernacular I would expect in a discussion about actual real-time (ie: deterministic, predictable, deadline based execution of tasks).

> Currently I do am controlling the accuracy of the solving time by bounding the call with a time recorder, but I then am doing it at a "moveit layer level", and was just suggesting a warning in case people believe solve is real time and don't control the amount of time, or an improvement to handle it internally at an ompl-layer level, which would enable all their planners to be called real-time ones (when called through solve(), so basically when used through MoveIt)...

Are you asking for clarification of the `allowed_planning_time` field in the planning request?

> or an improvement to handle it internally at an ompl-layer level, which would enable all their planners to be called real-time ones

I doubt that would be sufficient to be able to claim that OMPL is suitable for use in real-time contexts (nor whether the devs would want to claim something like that).

moucrob, 2019-02-15:
> Should I understand "countdown constraint" to be a deadline?

Yes

> "almost the same" is not really vernacular I would expect in a discussion about actual real-time (ie: deterministic, predictable, deadline based execution of tasks).

I agree it's not precise sorry, let's say a recorded time differing from 0.00001sec (10 microsec) from the inputed deadline. Sorry about the other words I know nothing in informatics.

> Are you asking for clarification of the `allowed_planning_time` field in the planning request?

No

> I doubt that would be sufficient to be able to claim that OMPL is suitable for use in real-time contexts (nor whether the devs would want to claim something like that).

Thank you

gavanderhoorn, 2019-02-15:
> > I doubt that would be sufficient to be able to claim that OMPL is suitable for use in real-time contexts (nor whether the devs would want to claim something like that).
> 
> Thank you

?

BryceStevenWilley, 2019-02-15:
I'm also a little confused here, by "realtime" do you mean an [anytime algorithm](https://en.wikipedia.org/wiki/Anytime_algorithm)?  

It seems that you are concerned about very small bits of extra compute after planning is done (`push_back` and `resize` of vectors is negligible compared to the overall time spent planning) . Any reason why?

moucrob, 2019-02-15:
> I'm also a little confused here, by "realtime" do you mean an [anytime algorithm](https://en.wikipedia.org/wiki/Anytime_algorithm)?

I would also love it to be "anytime" (I try to make it anytime as well in my wrapper), but no here I only speak about timeout constraint

> Any reason why?

To give an idea, if I use the input `timeout` being 0.5sec, some recorder tells me that sometimes it actually took  0.52sec, so I fear that replanning at 100Hz would not be possible when passing through the solve function